hawku/TabletDriver #315: How do you make it work with Veikk S640 please?

olcatt, 2018-12-24:
## Hardware/software information
Windows Version: win10 pro

Tablet Driver version: 2.3

Tablet model: Veikk s640

riley-badour, 2018-12-24:
Please read before you create a issue the solution is under the Installation the very fist step is there
https://github.com/hawku/TabletDriver/blob/master/README.md#installation

olcatt, 2018-12-24:
> Please read before you create a issue the solution is under the Installation the very first step is there
> https://github.com/hawku/TabletDriver/blob/master/README.md#installation

i did all those things it doesn't seem to work.
i used a previous version of this and it worked on my huion but it doesn't seem to work on my veikk
those are from an older pc btw so i think i might have missed something, probably the step one
sorry for the idiotic question

olcatt, 2018-12-24:
yeah i'm an idiot
i really missed the first step

closed 2018-12-24 by olcatt