hawku/TabletDriver #317: Tablet disconnecting itself

EXtremeExploit, 2018-12-25:
My cursor stops moving for 5 seconds every ~10 minutes, The tablet light is still on and it also makes the windows sound that a device has disconnected and never does the sound of a connection.

## Hardware/software information
Windows Version: Windows 10 Home Single Language 1803

Tablet Driver version: v0.2.3

Tablet model: CTL-490 Connected on a USB-Hub (pic below)
[image]

hawku, 2018-12-25:
@EXtremeExploit So you are saying that the tablet's USB connection drops and never reconnects?
How can the driver connect to a device that isn't detected by the Windows?
Try another USB cable, USB port or USB hub.

EXtremeExploit, 2018-12-25:
@hawku 
Windows USB connection drops for milliseconds, then tablet works again 10 seconds later.

I don't have any USB left, I have 3: Mouse, Keyboard and the USB Hub, I also tried to connect the tablet to a different USB but it seems to don't help.

By the way this wasn't happening after i updated to v0.2.3.

hawku, 2018-12-26:
@EXtremeExploit I still can't still see how this is a driver problem? It just reads data from the tablet and does not tell the tablet to disconnect.
There are multiple different reasons for USB to randomly disconnect . USB isn't usually very stable way to connect things to a PC.

Some possible reasons for USB disconnects:
- USB power saving (USB selective suspend)
- USB hub doesn't have enough power
- Bad USB cable
- Bad USB controller
- Faulty USB device

closed 2019-01-09 by hawku