hawku/TabletDriver #319: Anybody have any luck connecting a CTL-6100WL?

frylock567, 2018-12-26:
I'm adding in a custom config brought up by another user Deadspeck which was 

#
# Wacom CTL-6100WL
#
Tablet 0x056a 0x0378 0xFF0D 0x0001
Tablet 0x056a 0x0375 0xFF0D 0x0001
Name "Wacom CTL-6100WL"
ReportId 0x10
ReportLength 192
DetectMask 0x40
MaxX 21600
MaxY 13500
MaxPressure 4095
Width 216.0
Height 135.0
Type Wacom6100WL

However, I don't know if I'm typing it in wrong or whatever because it's not recognizing it. I tried copying and pasting this to the end of the config file as I assumed this is what you're supposed to do? Does it matter if I paste it before or after those 3 Hashtags at the end there? Or replace them entirely with the custom config? Not sure what I'm doing wrong. If there is no solution to this will there be future support for this model without the need of adding in a custom config? I'm more of "plug it in and it just works" kinda guy, I'm not really too good at this extra stuff. Any help would be greatly appreciated.

## Hardware/software information
Windows Version: Windows 10 Pro 64-Bit

Tablet Driver version: 0.2.3

Tablet model: CTL-6100WL

frylock567, 2018-12-26:
@hawku hey that works!! thanks so much. tablet buttons do work as well.

Ahin0223, 2020-08-25:
why i can't ?QQ    CTL-6100

Ahin0223, 2020-08-25:
TabletDriverGUI Not updated?