hawku/TabletDriver #320: Tablet connected but cannot sense input from tablet.

Fresstyler, 2018-12-27:
## Hardware/software information
Windows Version: Windows 10 Home

Tablet Driver version: 

Tablet model: Wacomb CTL 480S

Fresstyler, 2018-12-27:
[2018-12-27 11:12:49.231] \ Reading 'config/init.cfg'
[2018-12-27 11:12:49.231] >> Log startuplog.txt
[2018-12-27 11:12:49.232] Log file 'startuplog.txt' opened.
[2018-12-27 11:12:49.232] >> LogDirect true
[2018-12-27 11:12:49.232] Log direct print = True
[2018-12-27 11:12:49.232] >> Include "config\tablet.cfg"
[2018-12-27 11:12:49.232] \ Reading 'config\tablet.cfg'
[2018-12-27 11:12:49.232] >> #
[2018-12-27 11:12:49.232] >> # Example tablet definition:
[2018-12-27 11:12:49.232] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2018-12-27 11:12:49.232] >> #
[2018-12-27 11:12:49.232] >> # VID: 0x056a
[2018-12-27 11:12:49.232] >> # PID: 0x00dd
[2018-12-27 11:12:49.232] >> # HID Usage Page: 0x0D
[2018-12-27 11:12:49.232] >> # HID Usage: 0x01
[2018-12-27 11:12:49.232] >> #
[2018-12-27 11:12:49.232] >> #
[2018-12-27 11:12:49.232] >> # Wacom CTL-470 (Wacom drivers installed)
[2018-12-27 11:12:49.232] >> #
[2018-12-27 11:12:49.232] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2018-12-27 11:12:49.234] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2018-12-27 11:12:49.234] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2018-12-27 11:12:49.234] >> ReportId 0x02
[2018-12-27 11:12:49.234] >> ReportLength 11
[2018-12-27 11:12:49.234] >> DetectMask 0x40
[2018-12-27 11:12:49.234] >> MaxX 14720
[2018-12-27 11:12:49.234] >> MaxY 9200
[2018-12-27 11:12:49.234] >> MaxPressure 1023
[2018-12-27 11:12:49.234] >> Width 147.2
[2018-12-27 11:12:49.234] >> Height 92.0
[2018-12-27 11:12:49.234] >> TabletFormat WacomDrivers
[2018-12-27 11:12:49.234] >> #
[2018-12-27 11:12:49.234] >> # Wacom CTL-470 
[2018-12-27 11:12:49.234] >> #
[2018-12-27 11:12:49.234] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2018-12-27 11:12:49.236] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2018-12-27 11:12:49.236] >> Name "Wacom CTL-470"
[2018-12-27 11:12:49.236] >> ReportId 0x02
[2018-12-27 11:12:49.236] >> ReportLength 10
[2018-12-27 11:12:49.236] >> DetectMask 0x40
[2018-12-27 11:12:49.236] >> MaxX 14720
[2018-12-27 11:12:49.236] >> MaxY 9200
[2018-12-27 11:12:49.236] >> MaxPressure 1023
[2018-12-27 11:12:49.236] >> Width 147.2
[2018-12-27 11:12:49.236] >> Height 92.0
[2018-12-27 11:12:49.236] >> InitFeature 0x02 0x02
[2018-12-27 11:12:49.236] >> #
[2018-12-27 11:12:49.236] >> # Wacom CTH-470 (Wacom drivers installed)
[2018-12-27 11:12:49.236] >> #
[2018-12-27 11:12:49.236] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2018-12-27 11:12:49.237] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2018-12-27 11:12:49.237] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2018-12-27 11:12:49.237] >> ReportId 0x02
[2018-12-27 11:12:49.237] >> ReportLength 11
[2018-12-27 11:12:49.237] >> DetectMask 0x40
[2018-12-27 11:12:49.237] >> MaxX 14720
[2018-12-27 11:12:49.237] >> MaxY 9200
[2018-12-27 11:12:49.237] >> MaxPressure 1023
[2018-12-27 11:12:49.237] >> Width 147.2
[2018-12-27 11:12:49.237] >> Height 92.0
[2018-12-27 11:12:49.237] >> TabletFormat WacomDrivers
[2018-12-27 11:12:49.237] >> #
[2018-12-27 11:12:49.237] >> # Wacom CTH-470
[2018-12-27 11:12:49.238] >> #
[2018-12-27 11:12:49.238] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2018-12-27 11:12:49.239] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2018-12-27 11:12:49.239] >> Name "Wacom CTH-470"
[2018-12-27 11:12:49.239] >> ReportId 0x02
[2018-12-27 11:12:49.239] >> ReportLength 10
[2018-12-27 11:12:49.239] >> DetectMask 0x40
[... 287 lines cut ...]
[2018-12-27 11:12:49.264]   Report Id = 0x02
[2018-12-27 11:12:49.264]   Report Length = 10 bytes
[2018-12-27 11:12:49.264]   Detect Mask = 0x40
[2018-12-27 11:12:49.264]   Ignore Mask = 0x00
[2018-12-27 11:12:49.264]   Pen button map = '' '' '' 
[2018-12-27 11:12:49.264]   Aux button count = 0
[2018-12-27 11:12:49.264]   Init feature report: { 0x02, 0x02 }
[2018-12-27 11:12:49.264] 
[2018-12-27 11:12:49.265] Area:
[2018-12-27 11:12:49.265]   Desktop = 1920px x 1080px
[2018-12-27 11:12:49.265]   Screen Map = 1920px x 1080px @ X+0px, Y+0px
[2018-12-27 11:12:49.265]   Tablet =  100.00mm x 56.25mm @ Min(61.87mm, 34.42mm) Max(161.87mm, 90.67mm)
[2018-12-27 11:12:49.265]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2018-12-27 11:12:49.265] 
[2018-12-27 11:12:49.265] >> Start
[2018-12-27 11:12:49.266] TabletDriver started!
[2018-12-27 11:12:49.266] [STATUS] TABLET Wacom CTL-480
[2018-12-27 11:12:49.266] [STATUS] HID 056A 030E FF0D 0001
[2018-12-27 11:12:49.266] [STATUS] WIDTH 152.00000
[2018-12-27 11:12:49.266] [STATUS] HEIGHT 95.00000
[2018-12-27 11:12:49.266] [STATUS] MAX_X 15200
[2018-12-27 11:12:49.266] [STATUS] MAX_Y 9500
[2018-12-27 11:12:49.266] [STATUS] MAX_PRESSURE 1023
[2018-12-27 11:12:49.266] [STATUS] AUX_BUTTONS 0
[2018-12-27 11:12:49.266] >> Log Off

riley-badour, 2018-12-27:
Pleasw dont ever do that again just drag the startuplog.txt from the folder eith debug selected
You say your tablet is a 480s we have no support for that tablet yet s variant 
We just need your corporation and we can help and see if we surpport your tablet

Fresstyler, 2018-12-27:
Sorry .thought you guys would need it my mistake ...

Fresstyler, 2018-12-27:
Forgot to mention my tablet  was working with the previous version

Fresstyler, 2018-12-27:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2712587/startuplog.txt)

hawku, 2018-12-27:
@Fresstyler Open the console tab and enable the debug checkbox at the bottom.
After that it should print debug information about the pen movements to console.
Does it show any debug info?

closed 2019-01-09 by hawku

Fresstyler, 2019-02-04:
@hawku  sorry for the absence , you dont have to help me any more but yes there is debug info...

Fresstyler, 2019-02-04:
i still have the issue also my wacomb is installed