hawku/TabletDriver #321: Random black lines start appearing when i draw near a rectangular border.

SilentThorns, 2018-12-27:
## Hardware/software information
Windows Version: 

Tablet Driver version: TabletDriverV0.2.3

Tablet model: XP-PEN Star G640

[https://imgur.com/a/6TaRIiu](url)

hawku, 2018-12-27:
@SilentThorns Does that happen with the official drivers?
Does that only happen in middle of the tablet area?
I can't reproduce that with my G640, so it might be a hardware problem.

riley-badour, 2018-12-27:
I cannot reproduce that either on My g430s weird its like its giving a false x and y varible that lands at 0 great to have some some debuging info. Like startuplog.txt Also what happen swith stock drivers does it do the same.

riley-badour, 2018-12-28:
Its like hawku said it could be a hardware problem
That what it seems to be if it happens on stock drivers aswell.

closed 2019-01-02 by hawku

Devocub, 2019-09-26:
It's Paint.NET problem, but in latest version (4.2.4) it is fixed
[image]