hawku/TabletDriver #322: CTH 480 Recognition Data Mode (possible 200 rps)

asdfdsafa, 2018-12-29:
Is it possible to access the options to change the tablet to Recognition Data mode inside the driver? This would effectively make the Tablet run on 200rps instead of the default 133.

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

hawku, 2018-12-29:
@asdfdsafa Does the official drivers have this option?

asdfdsafa, 2018-12-29:
actually, they do, but for the wacom pro lines. It appears they locked it out of the drivers for the normal intuos, but the hardware should be supportive of it. But since the drivers are the same program for the pro line and the non pro line, it's only logical that the feature is hidden. I wished I had the programming skills to do something like this, and I feel kinda bad for not being able to contribute. I'm going to start learning c++ and c though,maybe in a month or so I'll be decent enough to start helping xD.

hawku, 2018-12-29:
@asdfdsafa Wacom Pro line does have completely different hardware.
Even if there were a option in the drivers to change the report rate, I don't think that the Wacom 480 firmware has any way to change the report rate.

asdfdsafa, 2018-12-30:
I don't know man :P if it works then jackpot; if not then, welp, whatever. I just know the option is hidden so it just makes me want to flip the switch and see what happens xD. If you got the time, try it dude, and don't forget to tell us what happened ;).

hawku, 2018-12-30:
@asdfdsafa There isn't any magical switch that can change the report rate of 480.
Option to switch from data recognition mode to normal mode is for 200 RPS tablets.
The purpose of that is to prevent problems with the slower computers.
133 RPS tablets don't have that option because the report rate is already low enough.

closed 2018-12-30 by hawku