hawku/TabletDriver #326: Wacom CTL-4100 (Wacom drivers installed) cant change tablet area??

Anderzs, 2018-12-31:
## Hardware/software information
Windows Version: 10

Tablet Driver version: v. 0.2.3

Tablet model: Wacom CTL-4100

The problem is that i click the button "Restart Driver" and it restarts and says "Connected to Wacom CTL-411 (Wacom drivers installed)" and thats fine right, but then i try and change the area and press save settings but nothing happens and i have the default area for my wacom tablet which is the whole tablet

I tried looking in the console but nothing suspicious seemed to pop up

hawku, 2018-12-31:
@Anderzs The problem is that you have Wacom drivers installed.
You can't use both drivers at the same time.

closed 2018-12-31 by hawku

Anderzs, 2018-12-31:
I cant possibly find any trace of any wacom drivers on my computer and it still tells me that i have Wacom drivers installed

Anderzs, 2018-12-31:
What files does the software look for when deciding if the tablet has drivers installed? 

Or how does it even look for it?