hawku/TabletDriver #330: Gliding more than usual

TimothyDong, 2019-01-02:
## Hardware/software information
Windows Version: Windows 10 Home 64-bit

Tablet Driver version: V0.2.3

Tablet model: Huion H640P

The cursor is gliding more than usual, I do not know if I cause the problem as I have input a command stating : "noise 20 0.2". cursor started to glide more than usual so I deleted the command but the problem still persisted. I also used the Anti-Smoothening filter at: Shape: 0.700 and Compensation: 25ms when I typed in the command

hawku, 2019-01-02:
@TimothyDong You shouldn't use noise reduction filter with H640P.
I assume that by gliding you mean hardware smoothing or input lag.
The gliding can't be completely removed, but the next driver release will have better anti-smoothing settings for H640P:
https://www.youtube.com/watch?v=z3sFsB2rG_s

TimothyDong, 2019-01-02:
I forgot to mention that I deleted everything and re-installed and the problem still continued

hawku, 2019-01-02:
@TimothyDong It's part of the tablet hardware, so deleting everything and re-installing doesn't help.

TimothyDong, 2019-01-02:
so do I have to wait for the next version of the driver to be released?

hawku, 2019-01-02:
@TimothyDong You can try the latest development version: http://hwk.fi/TabletDriver/TabletDriverDev_2019_01_02.zip

Enable the anti-smoothing filter and select Huion H640P preset.

TimothyDong, 2019-01-02:
Thank you, I'll check it out after I'm done with school

TimothyDong, 2019-01-03:
it still kind of drifts a bit when I abruptly stop my stylus, but I think the problem was my fault to begin with anyway, as I was screwing around with shit I didn't know

TimothyDong, 2019-01-03:
I don't know if this will work or is even the solution but maybe have a "Clear Command" button that will allow the tablet to "forget" the command that you have typed into the commands tab if this kind of thing were to happen again

hawku, 2019-01-03:
@TimothyDong Just remove commands from the commands tab, save settings and click restart driver.
`Noise 20 0.2` does the same as setting the noise reduction filter buffer to 20 and threshold to 0.2 mm.

closed 2019-01-03 by hawku