hawku/TabletDriver #331: Huion 1060 Plus not recognized

KTExT, 2019-01-02:
For whatever reasons It seems to not recognize it, It works well for my CTL480 and my XP-Pen G640 but It really doesn't wanna recognize my Huion 1060 Plus
I've tried installing vmulti only, or huion only (obviously doesn't work)
I tried installing as administrator (doesn't seem to install anything at all when I do that)
And I've tried installing Huion before vmulti

In the past with v0.1.5 I somehow managed to make it recognize my 1060 Plus but never got it to match the area properly so I've never really used it.

## Hardware/software information
Windows Version:  Windows 10 Pro 1803

Tablet Driver version: V0.2.3

Tablet model: New Huion 1060 Plus

hawku, 2019-01-02:
@KTExT Huion 1060 Plus is not supported by this driver.
You need to make a table configuration to `tablet.cfg`.

riley-badour, 2019-01-04:
look like this issue is solved but i would like to make sure something about our tablets  we need to know as my tablet has the image incorrectly printed by the looks of it we need you to help determine if its for all of our new 1060 8192
[image]

KTExT, 2019-01-04:
Yeah, the loose wire might explain why It doesn't wanna charge my phone/mouse properly (It's the original tablet cable I just use it for other stuff) 

here's the log

> [2019-01-04 13:53:50.964] Log file 'HuionStrings.txt' opened.
[2019-01-04 13:53:50.964] >> #Close tablet connection
[2019-01-04 13:53:50.965] >> CloseTablet
[2019-01-04 13:53:50.965] Tablet closed!
[2019-01-04 13:53:50.965] >> #Open new tablet connection and get Huion Strings
[2019-01-04 13:53:50.965] >> USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
[2019-01-04 13:53:50.969] WinUSB tablet found! Manufacturer = '', Product = '', Serial = ''
[2019-01-04 13:53:50.969] >> GetString 1 3
[2019-01-04 13:53:50.969] Requesting string ids from 1 to 3:
[2019-01-04 13:53:50.971]   String (1) = 'HUION'
[2019-01-04 13:53:50.972]   String (2) = 'PenTablet'
[2019-01-04 13:53:50.974]   String (3) = 'LuoYong70558'
[2019-01-04 13:53:50.974] >> GetString 100 104
[2019-01-04 13:53:50.974] Requesting string ids from 100 to 104:
[2019-01-04 13:53:50.976]   String (100) = '@¨ÿ '
[2019-01-04 13:53:50.978]   String (101) = ' '
[2019-01-04 13:53:50.980]   String (102) = 'HUION Animation Technology Co.,ltd'
[2019-01-04 13:53:50.982]   String (103) = '57635126-8'
[2019-01-04 13:53:50.983]   String (104) = '511121198107170558'
[2019-01-04 13:53:50.983] >> GetString 121 123
[2019-01-04 13:53:50.983] Requesting string ids from 121 to 123:
[2019-01-04 13:53:50.985]   String (121) = 'HA60-F400'
[2019-01-04 13:53:50.987]   String (122) = ''
[2019-01-04 13:53:50.989]   String (123) = 'HK On'
[2019-01-04 13:53:50.989] >> GetString 200 202
[2019-01-04 13:53:50.989] Requesting string ids from 200 to 202:
[2019-01-04 13:53:50.991]   String (200) = 'p'
[2019-01-04 13:53:50.992]   String (201) = 'HUION_T174_170519'
[2019-01-04 13:53:50.994]   String (202) = 'HUION Animation Technology Co.,ltd'
[2019-01-04 13:53:50.994] >> Log off

I also seem to hit the corner at roughly 16, I also noted that when I made the tablet work, I thought I was messing up the settings but I guess It's how the tablet is.

closed 2019-01-09 by hawku