hawku/TabletDriver #332: Weird mouse movement while playing Overwatch

MattiaPette, 2019-01-02:
Hello, while the TabletDriver app is in background, with or without my tablet plugged in, there are some weird issues in other games such as Overwatch and I have included a clip of it.
The cursor is probably resetting his position to the top left corner of the screen and makes the game unplayable.

I don't think is a major bug or concern because you can simply close the TabletDriver app and everything works fine again.

If anyone has the same issue in another game or can confirm this, it will be much appreciated because I can't test this in another PC.

I can provide any system information, just ask.

Link to the video: [Youtube](https://www.youtube.com/watch?v=NnqHn3HPRZk)

## Hardware/software information
**Windows Version:** 
Windows 10.0.17134.472
**Tablet Driver version:** 
0.2.3 (currently latest)
**Tablet model:** 
Wacom CTL-490

hawku, 2019-01-02:
@MattiaPette That shouldn't happen in the version 0.2.3
Have you enabled the debug checkbox in the bottom of the console tab?
That might cause that problem, because the driver will get information from every input device at startup when the debug checkbox is enabled.

MattiaPette, 2019-01-05:
Yeah you are right. The checkbox was the problem and removing it seems to do the trick. Now it's fine.
For the sake of testing i pulled the repo and checked out the master. Still works fine if you don't enable the debug checkbox.
Thanks 👍

closed 2019-01-09 by hawku