hawku/TabletDriver #333: Tablet Read Error after a few seconds.

Allxzh, 2019-01-03:
I installed everything according to the instructions and have restarted my PC. I don't have any other drivers installed currently but I can't really get it to work. I'm opening the TabletDriverGUI.exe in admin mode.

When I first plug in my tablet and start the drivers it recognizes my tablet and works for 5 seconds then I get a "Tablet Read Error!" and the driver stops working. Searched but didn't find another issue about it so thought that I might aswell make one myself.

[2019-01-03 17:56:52.760] >> Start
[2019-01-03 17:56:52.766] TabletDriver started!
[2019-01-03 17:56:52.766] [STATUS] TABLET Gaomon S56K
[2019-01-03 17:56:52.766] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2019-01-03 17:56:52.766] [STATUS] WIDTH 160.00000
[2019-01-03 17:56:52.766] [STATUS] HEIGHT 120.00000
[2019-01-03 17:56:52.766] [STATUS] MAX_X 25196
[2019-01-03 17:56:52.766] [STATUS] MAX_Y 18896
[2019-01-03 17:56:52.766] [STATUS] MAX_PRESSURE 2047
[2019-01-03 17:56:52.766] [STATUS] AUX_BUTTONS 0
[2019-01-03 17:56:52.766] >> Log Off
[2019-01-03 17:56:52.773] Log file 'startuplog.txt' closed.
[2019-01-03 17:56:52.773] >> LogDirect False
[2019-01-03 17:56:52.773] Log direct print = False
[2019-01-03 17:56:52.773] 
[2019-01-03 17:56:52.773] Driver started!
[2019-01-03 17:56:52.773] 
[2019-01-03 17:56:57.105] [ERROR] [TabletHandler] Tablet Read Error!
Driver stopped!

That is what shows up in my Console.

## Hardware/software information
Windows Version: Windows 10 Home

Tablet Driver version: v0.2.3

Tablet model: Gaomon S56k

hawku, 2019-01-03:
@Allxzh Are you sure that the USB connection doesn't drop out?
For example the USB port on my Gaomon S56K is very loose, so the USB will drop about every minute if i don't glue or tape the connector to the tablet.

Allxzh, 2019-01-03:
Holy shit you are so smart. I've never experienced this issue before but changing the cable and making sure it was fastened fixed it. Thank you.

closed 2019-01-03 by hawku