hawku/TabletDriver #335: Fix for "Driver stopped. Restarting!"

AzGround, 2019-01-06:
I had a problem with the driver when I ran TabletDriverGUI. He read the first line of config / tablet.cfg and immediately rebooted. Removing the driver from XP-PEN is not enough, because in Windows 10, when you connect a device, your driver is downloaded. To remove it, you need to go to Settings, Devices, find the G430S and click delete. After that, the program will see your tablet.

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

vekhem, 2020-04-26:
Где мне именно найти Settings > Devices и т.д.?

dDmioko, 2020-04-26:
@vekhem в параметрах Windows
[image]

vekhem, 2020-04-26:
Увы, но не помогло. Не знаю, что еще можно попробовать...

dDmioko, 2020-04-27:
@vekhem попробуй не отключая девайса сделать следущее:
1. Удалить TabletDriver
2. Удалить драйвер в устройствах Windows
3. Установить TabletDriver
Мб нужна перезагрузка Windows

vekhem, 2020-04-27:
Это тоже не помогло, но спасибо за помощь