hawku/TabletDriver #340: Very small amounts of latency

Navelol, 2019-01-13:
So I got my tablet to play Osu! and it didn't take long for my friends to recommend these tablet drivers.  When I first downloaded the drivers, they worked fine for about a month and 2 weeks.  One day I noticed that my pen started to feel a little bit behind the cursor on the screen.  This impacted my game play a lot and I've tried everything to fix, or at least find out why it's happening.  If I start up a game it works fine for about 5 seconds before having some random latency.  This also happens when hovering.  I've tried restarting the computer, changing ports for the tablet, changing Osu! settings, adding anti-smoothing, using different areas on my tablet, hovering, uninstalling and reinstalling the drivers, and even using different pen nibs.  Nothing seems to work.  I've even used the actual VEIKK S640 drivers and that seemed to fix it so I know that it's the hawku drivers.  The problem is, I like the hawku drivers a lot more than the veikk drivers so there's nothing I can really do.   

## Hardware/software information
Windows Version:  10 pro 64 bit

Tablet Driver version: v0.2.3

Tablet model:  VEIKK S640

Banbeucmas, 2019-01-13:
Try to check the: Only when pen button down box next tho the filter box

hawku, 2019-01-13:
@Navelol Have you enabled the smoothing filter?
Is the mode set to absolute?
You could also try the SendInput (Debug) mode.

You should record a debug log:
- Open the Console tab and right click the console output to open a context menu.
- Select "Start debugging" from that menu. It will start to record debug data to a `debug_*.txt` file in the driver's folder.
- Try to reproduce the latency problem while the debug log is running.
- Close the GUI when you are sure that you have captured the problem.
- Attach the `debug_*.txt` and `startuplog.txt` to a GitHub comment.

hawku, 2019-01-14:
@Navelol That debug log is made in Windows Ink mode. Can you make one in absolute mode?

From that debug log I can only see the normal hardware smoothing latency.
At the start of that debug log the tablet is in 250 RPS mode and the hardware smoothing is at the lowest point.
When the pen pressure is detected, the tablet switches to 166 RPS mode and the hardware smoothing latency is about 50% higher.
That hardware smoothing switching will also happen with the official VEIKK drivers.

Navelol, 2019-01-14:
Sorry about that!  It seems I didn't save my settings before I started the debug.  This time I used the "SendInput" option.

[debug_2019-01-14_16_47_52.txt](https://github.com/hawku/TabletDriver/files/2757200/debug_2019-01-14_16_47_52.txt)
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2757201/startuplog.txt)

I also ran a debug in the normal Absolute mode, if that makes any difference haha.

[debug_2019-01-14_16_52_27.txt](https://github.com/hawku/TabletDriver/files/2757208/debug_2019-01-14_16_52_27.txt)