hawku/TabletDriver #341: Star G640 not working on Hawku

Vaf167, 2019-01-15:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: Star G640

Hello everyone, im having serious issues with my tablet since i cant use hawku with the G640 the pen just dont moves when moving it. 
Any kind of help will be appreciated 
Regards, Vaf
[image]

hawku, 2019-01-15:
@Vaf167 Does it work with the official XP-Pen drivers?
Does the cursor move if you use other modes than Absolute?
Does it output tablet movement data to console if you enable the debug checkbox at the bottom?

Vaf167, 2019-01-15:
Hi! It does work with xpen drivers and dont know why but although relative works fine the debug mode works really well but when using absolute nothing occurs

hawku, 2019-01-15:
@Vaf167 You have wrong vmulti driver installed by the XP-Pen driver installer.
Uninstall and reinstall the vmulti driver.

closed 2019-01-25 by hawku