hawku/TabletDriver #343: TabletDriverService - app error

MugetsuGZ, 2019-01-16:
Hello! I have certain problem: after formatting and instaling TabletDriver, etc. I got TabletDriverService app error. It's 0xc000007b error. I have tried uninstalling Net Framework, VC Redistributable and TabletDriver, nothing helped. Here is screenshot of error, sorry for foreign language: https://puu.sh/CxIQS/3151419878.png

## Hardware/software information
Windows Version:  Windows 10 (1809)

Tablet Driver version: v0.2.3

Tablet model: Wacom CTL-471

hawku, 2019-01-16:
Did you install the correct C# .NET and C++ libraries?
https://github.com/hawku/TabletDriver#installation

MugetsuGZ, 2019-01-16:
:man_facepalming: Reinsatlling Visual C++ Redist helped... Thanks

closed 2019-01-16 by MugetsuGZ