hawku/TabletDriver #344: Huion H950P not working with drivers.

DrTrouserPlank, 2019-01-17:
## Hardware/software information
Windows Version:  10

Tablet Driver version:  Hawku 0.2.3

Tablet model: Huion H950P

when running the install as instructed and running the hawku driver GUI, it gives the error "connected to unkwown Huion tablet, check
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2768365/startuplog.txt)"

The tablet doesn't work which is strange since I think this is just an upscaled version of the 640P which apparently does work. Going into console and enabling debug, and then dragging the pen on the tablet causes lots of debug tablet read events so I assume it is actually reading the pen position.

Is there any way to make this work with my model?

riley-badour, 2019-01-17:
Yes try this. Also try the tablet buttons if it does work aswell and we will get your tablet supported

#
# Huion H950P 8192
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
CheckString 201 "HUION_T172"
Name "Huion H950P"
ReportId 0x08
ReportLength 12
DetectMask 0x80
IgnoreMask 0x60
MaxX 44196
MaxY 27432
MaxPressure 8191
Width 221
Height 138
InitStrings 100 200
AuxReportId 0x08
AuxReportLength 12
AuxDetectMask 0xE0
AuxgnoreMask 0x80
AuxCustomData ReportId Source=0
AuxCustomData ButtonsLow Source=4 SourceMask=0x3F
AuxCustomData Detect Source=1
AuxButtonCount 8

DrTrouserPlank, 2019-01-17:
> Yes try this. Also try the tablet buttons if it does work aswell and we will get your tablet supported
> 
> Sorry about formating im no where near a computer
> 
> #Huion H950P 8192
> 
> USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
> CheckString 201 "HUION_T172"
> Name "Huion H950P"
> ReportId 0x08
> ReportLength 12
> DetectMask 0x80
> IgnoreMask 0x60
> MaxX 44196
> MaxY 27432
> MaxPressure 8191
> Width 221
> Height 138
> InitStrings 100 200
> AuxReportId 0x08
> AuxReportLength 12
> AuxDetectMask 0xE0
> AuxgnoreMask 0x80
> AuxCustomData ReportId Source=0
> AuxCustomData ButtonsLow Source=4 SourceMask=0x3F
> AuxCustomData Detect Source=1
> AuxButtonCount 8

Yes that works, thanks.

riley-badour, 2019-01-17:
Ok when you test this later does it use the full range of the tablet left to right and up to down. To the markings on the tablet mainly the very edges marked on it and do the buttons work so we can get this in to the official list when you test this later.

DrTrouserPlank, 2019-01-17:
> Ok when you test this later does it use the full range of the tablet left to right and up to down. To the markings on the tablet mainly the very edges marked on it and do the buttons work so we can get this in to the official list when you test this later.

So the full area definitely works, in fact it does extend a little past the dot markings in the corners but I think this is also the case on the official Huion drivers so probably more a case of the dots being slightly off rather than a software problem

Pen buttons work. As for the tablet itself, buttons 1-6 are assignable and work, but buttons 7&8 do not work (they can be assigned, but don't function). I think the 640P has 6 buttons so maybe that's part of the reason that 7&8 don't work with how we currently have the 950P set up?

hawku, 2019-01-17:
@DrTrouserPlank Change the buttons data format from:
`AuxCustomData ButtonsLow Source=4 SourceMask=0x3F`
to
`AuxCustomData ButtonsLow Source=4`

DrTrouserPlank, 2019-01-17:
> @DrTrouserPlank Change the buttons data format from:
> `AuxCustomData ButtonsLow Source=4 SourceMask=0x3F`
> to
> `AuxCustomData ButtonsLow Source=4`

All buttons work correctly now :)

DrTrouserPlank, 2019-01-17:
I don't know if it matters, but what do the width and height numbers in the tablet config mean in relation to this tablet (221, 138)? if that's an actual measurement in mm then those aren't the dimensions of the 950P, they sound more like the dimensions of the 640P. Maybe it doesn't matter but I thought I ought to mention it.

NoahDTheHorse, 2020-11-19:
what do I do with the code that was sent?