hawku/TabletDriver #346: Eraser support in Windows Ink mode

JTF195, 2019-01-21:
If possible, it would be nice if TabletDriver supported erasing in Windows Ink mode
(unless this is somehow exclusive to the official Wacom drivers)

## Hardware/software information
Windows Version:
18317.rs_prerelease.190111-1556

Tablet Driver version: 
Commit https://github.com/hawku/TabletDriver/commit/0d8cc3ef7d25aedb471be832ead11c62daae7b20, compiled with VS2017
Wacom Driver 6.3.32-4

Tablet model: 
PTH-660

hawku, 2019-01-25:
@JTF195 Windows Ink mode does support eraser output, but tablet pen eraser end isn't mapped as Windows Ink eraser.

Reading the eraser input needs to be done differently on every tablet, so I need USB data from the tablet to be able to figure out how the eraser is presented in the data.

You should record a debug log that shows normal data and eraser data.
Instructions: https://github.com/hawku/TabletDriver/issues/340#issuecomment-453816344

JTF195, 2019-02-08:
Ok, here is a debug log where I've used only the eraser end of my pen:

[debug_2019-02-08_09_11_21.txt](https://github.com/hawku/TabletDriver/files/2845414/debug_2019-02-08_09_11_21.txt)

attania, 2019-03-12:
@hawku I also have a PTH-660, and I would love for there to be separate inputs (or at least a way to disable my eraser end) for my tablet; is there anything I can do to help you regarding giving you data?

KoleckOLP, 2023-12-30:
Would it be possible to have the Eraser work similarly to how it does it Wacom drivers? By that I mean just flipping the pen would select the eraser tool and then drawing with the eraser would erase.

I tried the windows ink eraser toggle and hold and neither seem to work the way I expect them on my CTH-480