hawku/TabletDriver #347: Every time i put up the pen from the tablet, cursor goes top left

Skopega, 2019-01-22:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: MTE-450

hawku, 2019-01-25:
@Skopega I need more information about the problem.
You should record a debug log: https://github.com/hawku/TabletDriver/issues/340#issuecomment-453816344

Post the `debug_*.txt` and `startuplog.txt` to here.

Skopega, 2019-01-25:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2796766/startuplog.txt)
[debug.txt](https://github.com/hawku/TabletDriver/files/2796768/debug.txt)

riley-badour, 2019-01-25:
Thats not right tablet that a t172 ????? Is it a wacom or a huion tablet your trying to get help with
I see the driver is picking up a huion one but i dont know what model it is.

Real question do you have more than one tablet conected to your pc

Skopega, 2019-01-25:
I have the only tablet connectd to my pc right now. Its Wacom Bamboo MTE-450.

hawku, 2019-01-25:
@riley-badour Did you download a wrong `startuplog.txt`?

riley-badour, 2019-01-25:
#
# Huion H610P v2
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
CheckString 201 "HUION_T172"
Name "Huion H610P v2"
ReportId 0x02
ReportLength 8
DetectMask 0x90
IgnoreMask 0x60
MaxX 50800
MaxY 31750
MaxPressure 8191
Width 254
Height 158.75

if you have a h610 try this i don't know if it will work but its should

riley-badour, 2019-01-25:
yes i did hawku

hawku, 2019-01-25:
@Skopega Edit MTE-450 config in the `wacom.cfg`.
Change `DetectMask 0x00` to `DetectMask 0x10` and restart the driver.

Skopega, 2019-01-25:
Yeah, thanks, it works!

closed 2019-01-25 by Skopega