hawku/TabletDriver #349: Very High pen tap latency

BurningFlare, 2019-01-25:
When I'm tapping the pen I get very high input lag; making the game unplayable

## Hardware/software information
Windows Version: 10

Tablet Driver version: TabletDriverV0.1.5

Tablet model: xp-pen g430

hawku, 2019-01-25:
@BurningFlare Does that happen with the official drivers?
Did you uninstall or shutdown the official drivers before starting the TabletDriver?

You are using older version of TabletDriver, but it shouldn't cause any extra lag.
You could try the latest version.

BurningFlare, 2019-01-30:
I just installed the latest version, and the latency seems to be ok. But is there a way to make mouse inputs release instantly when I move the pen above and away from the tablet? Right now tapping the pen down and putting it up will still be one mouse input, and I can only make another mouse input when the pen is above the the hovering area. My pen tapping never goes that high, so is there a way to make a mouse input happen when I touch the tablet, and make the mouse input stop when I take the pen above from the tablet?

hawku, 2019-01-30:
@BurningFlare Mouse 1 button should be released when you release the pen pressure.
There might be something wrong with your tablet. Maybe the pen nib is stuck, so the pen pressure doesn't get fully released when you lift the pen up.