hawku/TabletDriver #350: Hawku not working at all

Draxyn, 2019-01-25:
Switched to Hawku around a month or two ago, had a lot of problems to the point where it won't even stay connected to my tablet any more. I've tried pretty much everything I can think of and I'm stuck as to what the problem is.

## Hardware/software information
Windows Version: 10

Tablet Driver version:  V0.2.3

Tablet model: Huion H420

Draxyn, 2019-01-25:
> could be the vmuti driver not being installed with administrator privileges

Already tried that one I'm afraid

hawku, 2019-01-25:
@Draxyn You could also try to remove and reinstall the Huion drivers that are provided with the TabletDriver.

Draxyn, 2019-01-25:
@hawku Already tried that, doesn't change anything.

hawku, 2019-01-25:
@Draxyn Does the official Huion drivers work with your tablet? Can you change the tablet area or other settings with the official drivers?

hawku, 2019-01-25:
@Draxyn So is the problem that the TabletDriver connects to the tablet but doesn't stay connected?
Can you record a debug log and try to record the disconnection.
Debug log instructions: https://github.com/hawku/TabletDriver/issues/340#issuecomment-453816344

Post the `debug_*.log` and `startuplog.txt` to here.

Draxyn, 2019-01-25:
I don't think I'll be able to replicate the connecting, it only connects briefly on launch and very rarely connects later on. @hawku

hawku, 2019-01-25:
@Draxyn Post the `startuplog.txt`

Draxyn, 2019-01-25:
@hawku there is no startlog, it doesn't produce one. No clue what's happening with it

Draxyn, 2019-01-25:
@hawku It opens but only shows an error message about the VMulti drivers and possible fixes

hawku, 2019-01-25:
@Draxyn Are the Huion or other tablet drivers running in the background?
Did the VMulti install correctly without error messages?

Draxyn, 2019-01-25:
@hawku That's weird, I tried uninstalling it again and reinstalling and it seemed to make it work, however long it will work I don't know but I'll keep you updated. I don't have any other tablet drivers installed anyway

HerendraTJ, 2019-02-07:
Please leave the SS so we can help you

alexcher-im, 2019-09-21:
Have the same issue with xp-pen deco 01. Sometimes input freezes and sometimes turns into windows ink mode, console in gui with debug logging just stops flooding at all. If i click on "Restart Driver" it can help or driver says "Tablet init failed!", then i need to reconnect tablet and restart driver once more.