hawku/TabletDriver #351: Can't move cursor and driver thinks my tablet is 2:1

Pituophis, 2019-01-26:
Moving my pen doesn't move the cursor and it thinks my tablet is 2:1
[image]

## Hardware/software information
Windows Version: 10.0.17134 

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640s

riley-badour, 2019-01-27:
are you sure its not a bad cable mini b usb are quite bad for connections and it looks like it connects and disconnects instantly   can you try different cable and send  another start up log if your having problems still

Re1za, 2019-01-27:
Hello. I am Japanese and I'm not good at English, but please
I am also exactly like this phenomenon
The device uses CTL 480 The driver is erasing properly

Pituophis, 2019-01-27:
it's the g640.
 Problem was just one of my usb ports. Thanks