hawku/TabletDriver #353: G430S didn't work in absolute mode

developerfromua, 2019-01-27:
## Hardware/software information
Windows Version: 
10
Tablet Driver version: 
0.2.3
Tablet model: 
Xp-Pen G430S

Can't move cursor in absolute mode. Relative and Windows lnk works fine. Recently i connected second monitor and after this actions driver didnt't work. Monitor was unplugged but problem still exist. Tried reinstall hawku driver, unistall and install xp-pen driver(in last version xp-pen driver also didn't work absolute mode, but old drivers works perfectly). Also my tablet was connected in Linux system. What i have done: deleted all drivers, install vmulti, reboot windows but nothing happened. I don't know what i have to do.

developerfromua, 2019-01-27:
Sorry for some mistakes, i can't speak  English good

hawku, 2019-01-27:
@developerfromua You might have a wrong VMulti driver installed.
If you somehow can't uninstall the wrong VMulti driver, you could try to install the XP-Pen driver with a correct VMulti driver:
https://github.com/hawku/TabletDriver/wiki/FAQ#q-ive-installed-vmulti-driver-but-i-get-an-error-cant-open-vmulti-device

developerfromua, 2019-01-28:
> @developerfromua You might have a wrong VMulti driver installed.
> If you somehow can't uninstall the wrong VMulti driver, you could try to install the XP-Pen driver with a correct VMulti driver: [...]

didn't work
my tablet works only with xp-pen driver (star 4 series)
recently your driver worked good, but now i have no idea what to do
i think i may reinstall windows ;\

hawku, 2019-01-28:
@developerfromua You could try uninstall every "Pentablet HID" device from the device manager.
Select "Delete the driver software for this device" while uninstalling the device.

developerfromua, 2019-01-31:
> @developerfromua You could try uninstall every "Pentablet HID" device from the device manager.
> Select "Delete the driver software for this device" while uninstalling the device.

I will try this. Is your driver support 2nd monitor connection? I thought is was only for win7. What about win10?