hawku/TabletDriver #355: CTH-680 Driver V0.2.3, Tablet not found.

Rhizix, 2019-01-28:
I am able to run my Tablet on TabletDriverV0.1.5 just fine.
But when I start the V0.2.3 GUI/Driver it says there are no devices detected "Tablet not found!'.
Any ideas?

## Hardware/software information
Windows Version:W10 1809

Tablet Driver version: 0.2.3

Tablet model: CTH680

hawku, 2019-01-29:
@Rhizix Upload the `startuplog.txt` from versions 0.1.5 and 0.2.3 to here.

Rhizix, 2019-01-29:
[startuplog-V0.1.5.txt](https://github.com/hawku/TabletDriver/files/2808565/startuplog-V0.1.5.txt)
[startuplog-V0.2.3.txt](https://github.com/hawku/TabletDriver/files/2808566/startuplog-V0.2.3.txt)
I've attached both files

hawku, 2019-01-29:
@Rhizix Something weird is going on with the version 0.2.3

Latest stability fixes might have already fixed that problem, so you should test if the problems still occurs with the latest development version:
http://hwk.fi/TabletDriver/TabletDriverDev_2019_01_29.zip

hawku, 2019-01-29:
@Rhizix I checked your v0.2.3 `startuplog.txt again and it seems that you are running an old `tablet.cfg` with the version 0.2.3.

Rhizix, 2019-01-29:
I downloaded your zip and it works now, thanks for the rapid reponse :)
I think this can be closed now?