hawku/TabletDriver #361: Support for Huion 580

Grasscirno, 2019-02-02:
Hello, I own an old Huion 580 tablet and you can add support for it! 
Currently when I plugged it in with the latest TabletDriver version it gets wrongly detected as osu!tablet and I don't know how to change that. I tried opening the tablet.cfg file and change GetStrings 1 20 from the osu!tablet config to GetStrings 1 250 or GetStrings 190 230 as in this thread: https://github.com/hawku/TabletDriver/issues/255; but the tablet still got detected as osu!tablet. Hope you guys could help :)

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.4

Tablet model: Huion 580
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2824513/startuplog.txt)

riley-badour, 2019-02-02:
If you can type GetStrings 1 250. Into the console and than paste into a text file. Then drop it here i need that to identify your tablet.

riley-badour, 2019-02-02:
@hawku hunion tablets have at least 3 identifiers
String 6 and 201 and now 121 may be just another one. 
String (121) = 'M508' 
Could be a internal name for it. Or something i

Now for you grasscirno
I need some debug info there a checkbox. On the console page  click that. Now move your pen a bit on the tablet copy and put it to paste bin.

Grasscirno, 2019-02-02:
> @hawku hunion tablets have at least 3 identifiers
> String 6 and 201 and now 121 may be just another one.
> String (121) = 'M508'
> Could be a internal name for it. Or something i
>
> Now for you grasscirno [...]

https://pastebin.com/6TDvVHnc Here it is. Hope this helps!

Grasscirno, 2019-02-03:
Hello. I managed to get some more information on the tablet with uclogic-tools. Hope this help :)
https://pastebin.com/fnhw8upu