hawku/TabletDriver #363: MSVCP140.dll and VCRUNTIME140.dll not found when I launch tablet driver

codly68, 2019-02-05:
Every time i launch the driver it just continuously pops up with those two errors. I have no idea what to do

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

riley-badour, 2019-02-05:
Please at least read the install instructions its even the first step https://github.com/hawku/TabletDriver/blob/master/README.md#installation

codly68, 2019-02-05:
I did read the instructions, I had both things from the first step installed but redownloading one of them seems to have solved the problem

riley-badour, 2019-02-05:
Restart your computer always helps

codly68, 2019-02-05:
Well it seems to working now but Its giving the driver stopped restarting error.

riley-badour, 2019-02-05:
You should try to install the vmuti drivers with admin and drop a startuplog.txt

codly68, 2019-02-05:
Well all seems to be working just had to uninstall and reinstall the vmulti driver

closed 2019-02-05 by codly68