hawku/TabletDriver #364: Tablet connecting and sensing input but cursor does not move.

Fresstyler, 2019-02-05:
## Hardware/software information
Windows Version: Windows 10 Home

Tablet Driver version: 0.2.3

Tablet model: Wacom CTL 480

Fresstyler, 2019-02-05:
After un-installing my wacom drivers  my tablet connects to the hawku driver , senses input but my cursor shows no movement .

Fresstyler, 2019-02-06:
no, could you advise me on how to or link me to instructions  much appreciated

Fresstyler, 2019-02-06:
i have installed vmulti with a restart but now i cant get rid of the wacomb drivers . i can run both hawku drivers and wacom drivers at the same time.

Fresstyler, 2019-02-06:
in programmes and features , the wacom drivers are invisible and cant be seen in device manager...

qbuss, 2019-02-08:
1. uninstall wacom drivers form "programs and settings" and from "devices"
(try to do it while the tablet is unplugged)

2. Install vmulti

3. Check if your tablet works

4. profit

if it doesn't work then go to the recovery mode https://www.digitalcitizen.life/4-ways-boot-safe-mode-windows-10
and then do it from begin

Fresstyler, 2019-02-08:
right i ave gotten the wacom drivers to stop and i am able to run the software with my tablet it works great . but now i have to always go to task manager and end wacom processes every time i boot up any way to fix this? thanks for the help :D

riley-badour, 2019-02-08:
Uninstall the wacom software

qbuss, 2019-02-08:
@Fresstyler do things I've said