hawku/TabletDriver #368: screen mapping bug (dual monitor)

qbuss, 2019-02-08:
## Hardware/software information
Windows Version:  10 Pro

Tablet Driver version: every, even the A"TabletDriverDev_2019_01_13" version

Tablet model:  Veikk s640, xp-pen 640

Hi, the driver behaves weird if I have 2 screens (one 2k and other one 1080p)

So If I set my drivers like this
[image]
Then I can only use the red part of my screen
[image]
But If I set my drivers like this
[image]
Then I can use the whole 2k (bigger in the driver) screen but I can't use the 2nd screen.

Oh and I tried to set my custom desktop size, but that also doesn't work.

hawku, 2019-02-08:
@qbuss That is a problem with some hardware configurations. Do you have multiple graphics adapters?
Use the compatibility mode if you want to use both monitors or set the desktop size to same as the second monitor resolution.

qbuss, 2019-02-08:
[image]
Yes But I'm only using the Nvidia one, give me a hour and I'll disable the intel gpu

qbuss, 2019-02-08:
oh and also the windows ink and compatibility mode works on 2 monitors and those don't have the bug

qbuss, 2019-02-08:
oke I've disabled the intel gpu in bios (even uninstalled the device)
[image]

The bug is still there

hawku, 2019-02-09:
@qbuss Yeah, you just have to use the compatibility mode.
I haven't yet figured out a proper way to detect when Windows can't use absolute mouse positioning on every monitor.
Even if I figure that one out, the only thing I can do to it is to disable the standard absolute mode on those systems and force it to use the compatibility mode.

andyrkn, 2019-06-10:
@qbuss the workaround (for me) is to disable the smaller screen (start+p -> use second screen only) start the Tablet Driver and connect your tablet then minimize the driver and reactive the other smaller screen. After that you can continue using ur tablet as normal