hawku/TabletDriver #374: Driver stops working when using stylus in photo editing software

nriordan1130, 2019-02-19:
I edit photos using Affinity photo editing software, and whenever I use the stylus in that program, my Hawku tablet driver essentially malfunctions.  The tablet area changes itself and will not go back to its normal size unless I run the remove_vmulti_driver Batch file then reinstall it. Even then, it will only sometimes correct itself.

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640

SAO-Allan, 2019-02-21:
Is your driver set to windows ink mode?
If not try switching to it, best of luck.

InfinityGhost, 2019-02-25:
Sounds like more of an issue with the software taking over your tablet and using its own drivers instead. I know that with GIMP it would use its own area if you enabled it in the settings, so I'd check in the program options for anything disabling it, or just doing some extra research on that issue.
Trying to narrow it down, does this happen with the normal XP-Pen drivers?

nriordan1130, 2019-02-25:
Thank you for the reply! (I forgot to respond to the first reply) I have not tried with the normal XP-Pen drivers as I almost only use the tablet for Osu!. To be honest, it's not too much of a problem for me and using the photo editing software. I just figured the Hawku developers should be made aware if a potential bug.To be honest, it does not really bother me too much since I can just not use the tablet in the software like I've always done. However, I would like to note that the issue has happened even after the photo software is closed and Osu! Is not even open. Not sure how that one happened, but I managed to fix it by just uninstalling and reinstalling using the batch files in like 20 seconds. Again, it does not really effect me, but I just thought you guys should at least have knowledge of the situation. Thanks for the reply! Glad to see the community is as great as ever! Hope all goes well in the future.- Nick Riordan 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/hawku/TabletDriver/issues/374#issuecomment-467016907",
"url": "https://github.com/hawku/TabletDriver/issues/374#issuecomment-467016907",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]