hawku/TabletDriver #379: Can't install vmulti driver

6Digit, 2019-02-26:
## Hardware/software information
Windows Version: Windows 10 64-bit

Tablet Driver version: devocub edition v0.1.5.5

Tablet model: cth 490

InfinityGhost, 2019-02-26:
Provide some screenshots and logs if possible to help diagnose

6Digit, 2019-02-26:
> Provide some screenshots and logs if possible to help diagnose

will this do? : https://i.gyazo.com/57d96818729feaabc4d748a9edb6ee4d.gif?_ga=2.149430527.878875188.1551139790-1674263470.1551139790

InfinityGhost, 2019-02-26:
Odd, was it running as admin?

6Digit, 2019-02-26:
> Odd, was it running as admin?

i'm unsure

6Digit, 2019-02-26:
let me try again

InfinityGhost, 2019-02-26:
Run the vmulti driver installer as admin is what I meant

ItsLemonTime, 2019-02-26:
Heyo. I'm also having the same issue. The install window immediately disappears. If I run it as not-administrator, the window shows some shit, and then asks me for access, and then disappears.

6Digit, 2019-02-26:
> Run the vmulti driver installer as admin is what I meant

is it possible to install it when it is not in the folder?

6Digit, 2019-02-26:
still disappears

InfinityGhost, 2019-02-26:
Well really we've narrowed this down to an issue with the vmulti driver, and not TabletDriver

6Digit, 2019-02-26:
Is there anything that I can do for the time being in an attempt to fix it?

InfinityGhost, 2019-02-26:
At the moment, I have no clue besides trying to install it in compatibility mode or something

6Digit, 2019-02-26:
ok
thank you for your help, I will continue to wait for a fix but I guess I will have to go back to my old drivers

qbuss, 2019-03-01:
1. disconnect the tablet (so the windows won't install the drivers automaticly).
2. uninstall wacom drivers.
3. isntall hawku drivers as admin
4. open tabletdriver (make sure that it's unpacked in the same directory as all the other files)
so it looks like this.
[image]

Rangga676, 2020-07-05:
I ran the install_vmulti_64.bat in the driver_vmulti folder with admin and it works

Xxsh4d0wsxX, 2020-09-13:
i cant install it when i do it will spam CMD with this message

C:\WINDOWS\system32>REM  --> Check for permissions

C:\WINDOWS\system32>"C:\WINDOWS\system32\cacls.exe" "C:\WINDOWS\system32\config\system"   1>nul 2>&1

C:\WINDOWS\system32>REM --> If error flag set, we do not have admin.

C:\WINDOWS\system32>if '2' NEQ '0' (
echo Requesting administrative privileges...
 goto UACPrompt
)  else (goto gotAdmin  )
Requesting administrative privileges...

C:\WINDOWS\system32>echo Set UAC = CreateObject("Shell.Application")    1>"C:\Users\username~1\AppData\Local\Temp\getadmin.vbs"

C:\WINDOWS\system32>echo UAC.ShellExecute "C:\Users\username~1\DOWNLO~1\COMPRE~1\TABLET~1.3\DRIVER~3\INSTAL~1.BAT", "", "", "runas", 1    1>>"C:\Users\username~1\AppData\Local\Temp\getadmin.vbs"

someone plz help

WhySiska, 2021-02-15:
вылетает такая ошибка:
[image]