hawku/TabletDriver #380: Feature Request: Proper Huion 610 Pro Support

vexilli, 2019-02-27:
The driver actually does already work as is to some degree, however it treats the active area as if it was the 420 or something and I have to push it to one of the corners. I personally wouldn't put much priority in full button support, but if it could just detect the tablet for it's drawing area -- that'd be wonderful.

## Hardware/software information
Windows Version: Win 10 Home 1803 17134.590

Tablet Driver version: 0.2

Tablet model: H610PRO w/ P80 pen

InfinityGhost, 2019-03-08:
It's certainly possible to do if it already works to a degree, but we'd need to get some parameters from the tablet like maximum pressure, width, height, etc.

vexilli, 2019-03-09:
I can tell you that it's 10x 6.25" working area with 5080 LPI, 233RPS and 8192 levels of pressure sensitivity (as advertised to consumers), but if you'd need that in some kind of pixel coordinates or different information/ different format (actual raw input limits?), I don't know that I'd be able to provide much without assistance. I looked at the .ini and can't find anything there that'd obviously be useful since the area is defined as ratios (1 being full area). Let me know if there's anything I can provide given the know how though.

InfinityGhost, 2019-03-09:
Try running this dev build and see what it spits out in the console
[DevBuild.zip](https://github.com/hawku/TabletDriver/files/2948533/DevBuild.zip)
(if you dont trust my build, you can build it yourself with visual studio)

vexilli, 2019-03-09:
[console.txt](https://github.com/hawku/TabletDriver/files/2948757/console.txt)
[startuplog.txt](https://github.com/hawku/TabletDriver/files/2948758/startuplog.txt)

InfinityGhost, 2019-03-10:
So at the moment its detecting the tablet as the Huion osu! tablet because of the way the driver detects tablets, and we've got to get more information. Put `GetStrings 1 250` into the console on the new driver and report back with the logs

vexilli, 2019-03-10:
[console.txt](https://github.com/hawku/TabletDriver/files/2948958/console.txt)

InfinityGhost, 2019-03-10:
Try this one, tell me how it works and drop logs here again
[Debug.zip](https://github.com/hawku/TabletDriver/files/2948974/Debug.zip)

vexilli, 2019-03-10:
Upon launch I get the system errors as following: The code execution cannot proceed because MSVCP140D.dll not found, VCRUNTIME140D.dll not found, ucrtbased.dll was not found. and:
`ERROR! Can't start the driver!
  The operation has timed out.
Driver stopped!
` on console.

InfinityGhost, 2019-03-10:
No clue why you're just suddenly getting that now, try this
[Debug.zip](https://github.com/hawku/TabletDriver/files/2949017/Debug.zip)

vexilli, 2019-03-10:
Same 3 errors, however this time after dismissing them, they each reprompt once after which the executable stalls and proceeds to crash.

InfinityGhost, 2019-03-10:
Does the first devbuild.zip still work?

vexilli, 2019-03-10:
Yes.

InfinityGhost, 2019-03-10:
[Debug.zip](https://github.com/hawku/TabletDriver/files/2949025/Debug.zip)
this may work, it has to do with missing frameworks which i may have broke by accident lol

vexilli, 2019-03-10:
It behaved virtually identical to the last one with the errors and crashing.

InfinityGhost, 2019-03-10:
[Debug.zip](https://github.com/hawku/TabletDriver/files/2949031/Debug.zip)
okay, this SHOULD work now, if not I'm just gonna be confused

vexilli, 2019-03-10:
[console.txt](https://github.com/hawku/TabletDriver/files/2949041/console.txt)

InfinityGhost, 2019-03-10:
How does it seem to function?

vexilli, 2019-03-10:
Full area on forced proportions uses about ~45% of the actual working area from around the middle of the tablet. I couldn't get the pen to register anything but (0,0) on primary monitor or invisibly hover over secondary with a 2 monitor connected regardless of the screen mapping. Disconnecting that allowed for at least working cursor control.

InfinityGhost, 2019-03-10:
Try these commands all together
`ReportId 0x07`
`ReportLength 8`
`DetectMask 0x00`
`MaxX 41700`
`MaxY 26252`

InfinityGhost, 2019-03-10:
I don't have a solid way to get the tablet's precise values at the moment, but I'm sure @hawku has a better, more consistent way of getting it to work

vexilli, 2019-03-10:
It looks like the full area but the right edge of the display stops slightly before where the taskbar would be if it was placed to the right side of the display

InfinityGhost, 2019-03-10:
Did it fix the stuck cursor?

vexilli, 2019-03-10:
Yeah, it did.

vexilli, 2019-03-10:
I just tried it with MaxX 40000 MaxY 25000 and I think that's it. Thanks for the help so far.

InfinityGhost, 2019-03-10:
Anything else seem incorrect?

InfinityGhost, 2019-03-10:
[Release.zip](https://github.com/hawku/TabletDriver/files/2949070/Release.zip)
This one should add the tablet support now

vexilli, 2019-03-10:
Upon first launch I got a runtime library debug error for TabletDriverService.exe saying abort() has been called and window's device manager had to rerecognize and setup my device. After that I didn't have the error again. The cursor is stuck on the top edge of the display however and randomly seems to right click while hovering

InfinityGhost, 2019-03-10:
Forgot to change that in the version I sent, this one should be fixed
[Release.zip](https://github.com/hawku/TabletDriver/files/2949616/Release.zip)

vexilli, 2019-03-10:
Fixed and stable from what I can tell.

harryxing, 2020-05-10:
how do I get this to work?