hawku/TabletDriver #384: bin/tabletdriverservice.exe not found!

jangglz, 2019-03-08:
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: H420

jangglz, 2019-03-08:
I have reinstalled multiple times, uninstalled ever other tablet driver i could find, and this still pops up every time I open GUI. Worked yesterday after this messaged popped up but mysteriously worked eventually.

InfinityGhost, 2019-03-08:
This may be a bit far-fetched, but do you run the GUI from a shortcut?

jangglz, 2019-03-09:
No, I go all the way to Temp

InfinityGhost, 2019-03-09:
You're running the program from the temporary folder? I'd try extracting it elsewhere and running it from the different location

jangglz, 2019-03-09:
I have but it doesn't change the error. Sometimes it wants to work and
sometimes it doesnt. idk

jangglz, 2019-03-09:
I have to reinstall the huion64 file everytime tho which im okay with but the 50/50 of it not letting me do anything just a little annoying :v

jangglz, 2019-03-09:
its amazing tho way better than the driver the tablet came with

Anonymous3785, 2019-05-13:
Have you found a Solution as of now? My Huion 420 presents the same problem.

InfinityGhost, 2019-05-14:
Well honestly try a fresh copy directly into a non-volatile folder like your downloads folder