hawku/TabletDriver #387: CTL-472 problem

Akiro-kun, 2019-03-10:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: v0.2.3

Tablet model: Wacom CTL-472

Problem: When I'm mapping or doing anything else on drivers, tablet settings don't change. (I have uninstalled wacom drivers) I downloaded these drivers in december, and problem was the same, but one time they were working. Thank you in advance for your help.

Akiro-kun, 2019-03-17:
Great community, thx.

mty22, 2019-03-19:
> Great community, thx.

Indeed! That's the beauty of open source.. so feel free to fix it and open a pull request. :)

Banbeucmas, 2019-03-19:
> Indeed! That's the beauty of open source.. so feel free to fix it and open a pull request. :)

Sacarsm aside. Let just enlighten him how things work if he wants software developer to fix his stuff. We all do suffer through this, but yeah, just keep an open mind and disregard the dumbness of the end user not providing enough details.

- What does console print when you plugged the tablet? Is it nothing or else?
- Had there anything you have done that day whatsoever before/after you saw the tablet work?
- Have you ever tried to mess up something like hard editing the config to see if you can change the settings by doing that way?
- How did you install the driver?

Akiro-kun, 2019-03-20:
> > Indeed! That's the beauty of open source.. so feel free to fix it and open a pull request. :)
> 
> Sacarsm aside. Let just enlighten him how things work if he wants software developer to fix his stuff. We all do suffer through this, but yeah, just keep an open mind and disregard the dumbness of the end user not providing enough details.
> 
> * What does console print when you plugged the tablet? Is it nothing or else?
> * Had there anything you have done that day whatsoever before/after you saw the tablet work?
> * Have you ever tried to mess up something like hard editing the config to see if you can change the settings by doing that way?
> * How did you install the driver?
> 
> Just to be clear, I also own a CTL472 and I didn’t see any problem with it.

1. [https://imgur.com/WyEoPzC](url) I don't see anything strange in there, but I haven't any knwledge about that. :p 
2. I don't remember, but I think that I have done everything like now.
3. No, I haven't.
4. I just used a download link on code part of this page, I was following an instruction and nothing changes.

P.S.
Sorry for any mistakes in this comment, I'm not the best in English, but I know it good.

Akiro-kun, 2019-03-26:
Please, answer

Syntractrix, 2019-08-28:
You have the wacom drivers installed, try to uninstall them completely through device manager and any other means (registry) ehem if it doesnt show up in uninstall panel then try to completely wipe it through the registry

closed 2024-06-20 by DanielYourSensei