hawku/TabletDriver #390: проблема с драйвером,не может найти файл

Freeq223, 2019-03-17:
eng;
incomprehensible for me an error that appeared unexpectedly for me, the driver starts, but the rn itself does not work, 
rus;
непонятная для меня ошибка которая появилась неожиданно для меня,драйвер запускается,но сам гп не работает,
[image]
[image]

помогите,я без понятия как это исправить

## Hardware/software information
Windows Version: 10

Tablet Driver version: v0.2.3

Tablet model: xp-pen g649

YChvanov, 2019-03-17:
У тебя в пути кириллица - перемести в корень диска, например

InfinityGhost, 2019-03-17:
If possible, could you get the error in English?
Если возможно, вы могли бы получить ошибку на английском языке?

YChvanov, 2019-03-17:
Same as this but with Сyrillic in path
[image]

InfinityGhost, 2019-03-17:
So the issue is that the file itself is nonexistent, but does it actually exist? Just try getting the newest release, extract it at like C:\root\ and run that uninstall.bat. Should uninstall it for sure that way.

Ponywka, 2019-03-22:
Только Украинцы и Русские испытывают такую проблему :D
Рекомендую переустановить систему, но при этом устанавливать не с Украинского/Русского образа, а с Английского(Американского / Межнационального)