hawku/TabletDriver #391: Wacom CTL480 tablet buttons support

rowario, 2019-03-22:
AuxHID 0x056a 0x030e 0xFF00 0x0080
AuxReportId 0x02
AuxReportLength 64
AuxDetectMask 0x80
AuxIgnoreMask 0x81
AuxCustomData ReportId Source=0
AuxCustomData ButtonsLow Source=3 SourceMask=0x0F
AuxCustomData Detect Source=2
AuxButtonCount 4

you need to add this lines in /config/tablet.cfg
https://i.imgur.com/gMw3HQd.png

InfinityGhost, 2019-03-23:
CTL480 button support has already been added in the dev build, so this issue can be closed.

InfinityGhost, 2019-03-23:
Also, why wouldn't you just add this as a PR if you already know what to change?

Banbeucmas, 2019-03-23:
> Also, why wouldn't you just add this as a PR if you already know what to change?

Some people doesn't know how to PR to be frank.

closed 2019-03-23 by rowario