hawku/TabletDriver #393: Best settings for XP-Pen G640S for osu?

KaetoKit, 2019-03-25:
What are the best settings for the XP-Pen G640S with this driver? Do I turn on Anti-smoothing Filter or Noise Reduction Filter?

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: XP-Pen G640S

InfinityGhost, 2019-03-25:
That comes down to preference in all honesty, but generally the lowest latency is the best to-go for any player. Adding the anti-smoothing filter or the noise reduction adds additional latency, but helps in the case you have a shaky cursor or just prefer smoother movements.

KaetoKit, 2019-03-25:
> That comes down to preference in all honesty, but generally the lowest latency is the best to-go for any player. Adding the anti-smoothing filter or the noise reduction adds additional latency, but helps in the case you have a shaky cursor or just prefer smoother movements.

So would the best to-go be no filters? That's what I have right now :) I want the lowest latency.

InfinityGhost, 2019-03-25:
Absolutely

KaetoKit, 2019-03-25:
Okay, thank you! ^^