hawku/TabletDriver #397: Xp-Pen G640S Anti-Smoothing Issue

ACrawford00, 2019-03-29:
I have the G640S and it has built in smoothing, and I absolutely can’t stand the delay at all. When I tried using the anti-smoothing filter, it got rid of the delay but has serious positioning issues. It says for this specific tablet to try enabling the noise reduction filter, but that literally made the problem worse. This is a video of what is happening, but for now I have to use my G640 until this is fixed. 

https://youtu.be/snulNV3ch5w

## Hardware/software information
Windows Version: 0.2.3

Tablet Driver version: 

Tablet model:

blackender89, 2019-04-09:
I have the same problem.Did you solve it?

ACrawford00, 2019-04-09:
No. I just use my G640 with the 640S pen. Pretty much perfect setup. It’s
hardware smoothing in the tablet, so idk how to fix it. I just use my
better tablet.

BaconPug, 2019-06-05:
having similar problems, hope they can do something about it. Also using G640 with my G640S pen

hawku, 2019-06-05:
Anti-smoothing filter can't fix all of the XP-Pen's hardware problems, but can reduce the latency.

XP-Pen's hardware smoothing and noise reduction algorithm is very poorly made, so it is very hard to predict the real pen position from the smoothed out data.
At lower speeds the tablet will just stop sending the position, while the pen is still moving.
Tablet will also stop sending the cursor position for about 50 milliseconds when you press or release a pen button.

BaconPug, 2019-06-05:
> Anti-smoothing filter can't fix all of the XP-Pen's hardware problems, but can reduce the latency.
>
> XP-Pen's hardware smoothing and noise reduction algorithm is very poorly made, so it is very hard [...]

Thanks for explaining.So according to what you’ve said, there is no way to fix this, right?

hawku, 2019-06-05:
@BaconPug Currently there is no way to fix those hardware problems.
You could ask XP-Pen support about those problems.

InfinityGhost, 2019-06-05:
Wow hawku's back

ACrawford00, 2019-06-08:
About time lol. For real tho, he needs a patron or something we can support
him at. Any news if previous issue is fixed or not?