hawku/TabletDriver #401: Xp-pen star 05 dont work with new version driver

Sertask, 2019-04-09:
My custom config:
#
# XP-Pen Star 05
#
Tablet 0x28BD 0x0071 0xFF0A 0x0001
Name "XP-Pen Star 05"
ReportId 0x02
ReportLength 8
DetectMask 0x80
MaxX 40640
MaxY 25400
MaxPressure 2047
Width 203.2
Height 127
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00

## Hardware/software information
Windows Version: 10 1803

Tablet Driver version: 0.1.5.5 (work with custom config), 0.2.3 (don't work)

Tablet model: Xp-pen star 05

Please add support

InfinityGhost, 2019-04-11:
We need some log files to fix this issue

Sertask, 2019-04-11:
[startuplog (1.5.5)(work).txt](https://github.com/hawku/TabletDriver/files/3068950/startuplog.1.5.5.work.txt)
[startuplog(2.3 old config and old bin file)(work).txt](https://github.com/hawku/TabletDriver/files/3068951/startuplog.2.3.old.config.and.old.bin.file.work.txt)
[startuplog(2.3 old config and new bin file)(dont work).txt](https://github.com/hawku/TabletDriver/files/3068953/startuplog.2.3.old.config.and.new.bin.file.dont.work.txt)
[startuplog(2.3 new config and new bin file)(dont work).txt](https://github.com/hawku/TabletDriver/files/3068954/startuplog.2.3.new.config.and.new.bin.file.dont.work.txt)
bin file - TabletDriverService.exe