hawku/TabletDriver #402: GUI crashes when trying to open, shows as suspended then closes in Task Manager

Cecelia22, 2019-04-10:
## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: CTL-480

Have reinstalled a number of times and to no prevail. Driver was working fine yesterday and now each time I try to start it, it opens for a few seconds ONLY in task manager then shows "Suspended" and then closes. Any way this is fixable?

InfinityGhost, 2019-04-11:
Log files are helpful for debugging this kind of situation

Cecelia22, 2019-04-11:
> Log files are helpful for debugging this kind of situation

How do I access the log? I have no file that shows anything having to do with the startup log

Cecelia22, 2019-04-12:
I think this is the log file you were looking for?

Log Name:      Application
Source:        Application Error
Date:          4/11/2019 7:40:53 PM
Event ID:      1005
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-H3NSAUV
Description:
Windows cannot access the file C:\Windows\Microsoft.NET\assembly\GAC_32\PresentationCore\v4.0_4.0.0.0__31bf3856ad364e35\PresentationCore.dll for one of the following reasons: there is a problem with the network connection, the disk that the file is stored on, or the storage drivers installed on this computer; or the disk is missing. Windows closed the program Update because of this error.

Program: Update
File: C:\Windows\Microsoft.NET\assembly\GAC_32\PresentationCore\v4.0_4.0.0.0__31bf3856ad364e35\PresentationCore.dll

The error value is listed in the Additional Data section.
User Action
1. Open the file again. This situation might be a temporary problem that corrects itself when the program runs again.
2. If the file still cannot be accessed and
	- It is on the network, your network administrator should verify that there is not a problem with the network and that the server can be contacted.
	- It is on a removable disk, for example, a floppy disk or CD-ROM, verify that the disk is fully inserted into the computer.
3. Check and repair the file system by running CHKDSK. To run CHKDSK, click Start, click Run, type CMD, and then click OK. At the command prompt, type CHKDSK /F, and then press ENTER.
4. If the problem persists, restore the file from a backup copy.
5. Determine whether other files on the same disk can be opened. If not, the disk might be damaged. If it is a hard disk, contact your administrator or computer hardware vendor for further assistance.

Additional Data
Error value: C0000242
Disk type: 3
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="49152">1005</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2019-04-12T00:40:53.066648700Z" />
    <EventRecordID>488</EventRecordID>
    <Channel>Application</Channel>
    <Computer>DESKTOP-H3NSAUV</Computer>
    <Security />
  </System>
  <EventData>
    <Data>C:\Windows\Microsoft.NET\assembly\GAC_32\PresentationCore\v4.0_4.0.0.0__31bf3856ad364e35\PresentationCore.dll</Data>
    <Data>Update</Data>
    <Data>C0000242</Data>
    <Data>3</Data>
  </EventData>
</Event>