hawku/TabletDriver #403: Huion HS64 support?

nekoellie, 2019-04-11:
Just a simple question (I hope)
Will there be support for the Huion HS64? The outside looks exactly the same as the H430p, but I can't say the same about how it works. My tablet will arrive in two weeks so if anyone needs any data from it, tell me how I can provide you with such.

InfinityGhost, 2019-04-11:
If it isn't already supported, support is quite easy to add for Huion tablets usually

nekoellie, 2019-05-15:
tried and can see that the HS64 does not work with TabletDriver yet. TabletDriver does see it tho. could add console log later today when i'm home

Jet-chann, 2019-08-18:
I'd like to see support on this tablet aswell.

InfinityGhost, 2019-08-18:
@Jet-chann I would like to try to see what I can get working on your end. I worked with @nekoellie on this previously and got to the point it detected the tablet and all, but windows ink always took priority.

closed 2020-04-10 by nekoellie