hawku/TabletDriver #404: CTH-480 DRIVERS NOT WORKING

jadenkim89, 2019-04-12:
I honestly have no idea why but even when I restarted my computer and disabled windows ink it defaults to some other driver rather than the hawku drivers. The reason I can tell is that it uses my full tablet area and I don't have it set to that. So idk if this is enough info to help somebody help me but thanks!

## Hardware/software information
Windows Version: Windows 10 Home

Tablet Driver version: All driver version that I tested (2.0, 2.3)

Tablet model: Wacom CTH-480

InfinityGhost, 2019-04-18:
More than likely, it is the wacom drivers that are taking over here. Uninstall those drivers or use the `.bat` file in the directory that disables the wacom drivers.

dumbstersky, 2019-06-01:
it's happening to we as well i tried doing all of the possible fixes and made sure my wacom drivers were gone but it wont work