hawku/TabletDriver #407: Can't start driver check console!

Gruee, 2019-04-19:
I have no idea why this is happening... My tablet isnt working with the driver and cant do anything with it. Also i cant save any of my layouts and stuff it says: Error occured while saving configuration. Please help me cuz i cant play osu :(

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: Intuos CTH 480
[image]

InfinityGhost, 2019-04-19:
Well what does the console say

Gruee, 2019-04-19:
ugh never mind im just dumb and extracted the program from the folder that seems to have caused the problem... A mod can close this thread since the issue has been resolved :)

InfinityGhost, 2019-04-19:
(you can click close issue to do that yourself)

closed 2019-04-19 by Gruee