hawku/TabletDriver #411: Please update to support Gaomon S620

KarenAraragi, 2019-04-29:
I would love to use Hawku drivers for my Gaomon S620, but it doesn't seem that it is supported yet. Please look into this. <3 

## Hardware/software information
Windows Version: Windows 10 build 1809 

Tablet Driver version: Latest

Tablet model: Gaomon S620

InfinityGhost, 2019-04-29:
https://github.com/hawku/TabletDriver/pull/412 adds support for this tablet.

KarenAraragi, 2019-04-29:
@InfinityGhost Thank you!

closed 2019-04-29 by KarenAraragi

InfinityGhost, 2019-04-29:
[**Download this release here**](https://github.com/InfinityGhost/TabletDriver/releases/tag/v0.2.4-pre)
This is my private pre-release that includes the support for this tablet. This repository seems to have died and I want to keep it alive as long as possible, so I'll be improving on it over time.