hawku/TabletDriver #413: Gaomon S620 support

Voxee, 2019-05-02:
## Hardware/software information
Windows Version: 10 64 bit

Tablet Driver version: -

Tablet model: Gaomon S620

InfinityGhost, 2019-05-02:
Pull request https://github.com/hawku/TabletDriver/pull/412 adds support for the Gaomon S620.

InfinityGhost, 2019-05-02:
This repository seems to have died off, but I've been including the pull requests in my fork. I have a release with the support for the Gaomon S620 [here](https://github.com/InfinityGhost/TabletDriver/releases/tag/v0.2.4-pre)

Voxee, 2019-05-03:
ty for that

closed 2019-05-03 by Voxee

henrik801, 2019-08-17:
My tablet stops working when i open the Driver.  (Gaomon s620)
[2019-08-17 16:05:47.779] \ Reading 'config/init.cfg'
[2019-08-17 16:05:47.779] >> Log startuplog.txt
[2019-08-17 16:05:47.779] Log file 'startuplog.txt' opened.
[2019-08-17 16:05:47.779] >> LogDirect true
[2019-08-17 16:05:47.779] Log direct print = True
[2019-08-17 16:05:47.779] >> Include "config\tablet.cfg"
[2019-08-17 16:05:47.779] \ Reading 'config\tablet.cfg'
[2019-08-17 16:05:47.779] >> #
[2019-08-17 16:05:47.779] >> # Example tablet definition:
[2019-08-17 16:05:47.779] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 16:05:47.779] >> #
[2019-08-17 16:05:47.779] >> # VID: 0x056a
[2019-08-17 16:05:47.779] >> # PID: 0x00dd
[2019-08-17 16:05:47.780] >> # HID Usage Page: 0x0D
[2019-08-17 16:05:47.780] >> # HID Usage: 0x01
[2019-08-17 16:05:47.780] >> #
[2019-08-17 16:05:47.780] >> #
[2019-08-17 16:05:47.780] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-08-17 16:05:47.780] >> #
[2019-08-17 16:05:47.780] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-08-17 16:05:47.782] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 16:05:47.782] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-08-17 16:05:47.782] >> ReportId 0x02
[2019-08-17 16:05:47.782] >> ReportLength 11
[2019-08-17 16:05:47.782] >> DetectMask 0x40
[2019-08-17 16:05:47.782] >> MaxX 14720
[2019-08-17 16:05:47.782] >> MaxY 9200
[2019-08-17 16:05:47.782] >> MaxPressure 1023
[2019-08-17 16:05:47.782] >> Width 147.2
[2019-08-17 16:05:47.782] >> Height 92.0
[2019-08-17 16:05:47.782] >> TabletFormat WacomDrivers
[2019-08-17 16:05:47.782] >> #
[2019-08-17 16:05:47.782] >> # Wacom CTL-470 
[2019-08-17 16:05:47.782] >> #
[2019-08-17 16:05:47.782] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-08-17 16:05:47.784] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-17 16:05:47.784] >> Name "Wacom CTL-470"
[2019-08-17 16:05:47.784] >> ReportId 0x02
[2019-08-17 16:05:47.784] >> ReportLength 10
[2019-08-17 16:05:47.784] >> DetectMask 0x40
[2019-08-17 16:05:47.784] >> MaxX 14720
[2019-08-17 16:05:47.784] >> MaxY 9200
[2019-08-17 16:05:47.784] >> MaxPressure 1023
[2019-08-17 16:05:47.784] >> Width 147.2
[2019-08-17 16:05:47.785] >> Height 92.0
[2019-08-17 16:05:47.785] >> InitFeature 0x02 0x02
[2019-08-17 16:05:47.785] >> #
[2019-08-17 16:05:47.785] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-08-17 16:05:47.785] >> #
[2019-08-17 16:05:47.785] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-08-17 16:05:47.787] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-08-17 16:05:47.787] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-08-17 16:05:47.787] >> ReportId 0x02
[2019-08-17 16:05:47.787] >> ReportLength 11
[2019-08-17 16:05:47.787] >> DetectMask 0x40
[2019-08-17 16:05:47.787] >> MaxX 14720
[2019-08-17 16:05:47.787] >> MaxY 9200
[2019-08-17 16:05:47.787] >> MaxPressure 1023
[2019-08-17 16:05:47.787] >> Width 147.2
[2019-08-17 16:05:47.787] >> Height 92.0
[2019-08-17 16:05:47.787] >> TabletFormat WacomDrivers
[2019-08-17 16:05:47.787] >> #
[2019-08-17 16:05:47.787] >> # Wacom CTH-470
[2019-08-17 16:05:47.787] >> #
[2019-08-17 16:05:47.787] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2019-08-17 16:05:47.789] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-08-17 16:05:47.789] >> Name "Wacom CTH-470"
[2019-08-17 16:05:47.789] >> ReportId 0x02
[2019-08-17 16:05:47.789] >> ReportLength 10
[... 1205 lines cut ...]
[2019-08-17 16:06:38.576]   Report Id = 0x07
[2019-08-17 16:06:38.576]   Report Length = 8 bytes
[2019-08-17 16:06:38.576]   Detect Mask = 0x80
[2019-08-17 16:06:38.576]   Ignore Mask = 0x00
[2019-08-17 16:06:38.577]   Pen button map = 'MOUSE1' 'MOUSE2' 'MOUSE3' 
[2019-08-17 16:06:38.577]   Aux button count = 0
[2019-08-17 16:06:38.577]   Init string ids: 100 200 
[2019-08-17 16:06:38.577] 
[2019-08-17 16:06:38.577] Area:
[2019-08-17 16:06:38.577]   Desktop = 3840px x 1080px
[2019-08-17 16:06:38.577]   Screen Map = 1920px x 1080px @ X+1920px, Y+0px
[2019-08-17 16:06:38.577]   Tablet =  60.00mm x 33.75mm @ Min(30.00mm, 16.88mm) Max(90.00mm, 50.63mm)
[2019-08-17 16:06:38.577]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-08-17 16:06:38.577] 
[2019-08-17 16:06:38.577] >> Start
[2019-08-17 16:06:38.721] TabletDriver started!
[2019-08-17 16:06:38.721] [STATUS] TABLET Unknown Huion Tablet! Check startuplog.txt!
[2019-08-17 16:06:38.721] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2019-08-17 16:06:38.721] [STATUS] WIDTH 100.00000
[2019-08-17 16:06:38.721] [STATUS] HEIGHT 100.00000
[2019-08-17 16:06:38.721] [STATUS] MAX_X 10000
[2019-08-17 16:06:38.721] [STATUS] MAX_Y 10000
[2019-08-17 16:06:38.721] [STATUS] MAX_PRESSURE 2047
[2019-08-17 16:06:38.721] [STATUS] AUX_BUTTONS 0
[2019-08-17 16:06:38.721] >> Log Off

InfinityGhost, 2019-08-18:
@henrik801 If you need immediate tablet support, refer to the repository above. I have a working configuration for it.

Berkkyyyy, 2019-10-26:
i downloaded it but the driver is always crashing