hawku/TabletDriver #418: CTL-490 feels like driver is being ignored

Kottowo, 2019-05-18:
So, i started my pc, wanting to play some osu, when i started it up, my tablet area was reset to full tablet area. Windows ink popped up, disabled it but it didn't affect anything. When i close the driver however, the tablet still works. I've tried disabling the wacom drivers (which i didn't install since this was a new pc and didn't need to since this driver exists), the huion drivers (just to make sure, even tho i use wacom), uninstalled and re installed the Vmulti drivers and messing with the settings in the driver itself. When i messed with the settings, nothing changed so i have a feeling something else is denying the tablet driver and it's using that instead. Hopefully someone can help...

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.2.3

Tablet model: CTL-490

InfinityGhost, 2019-05-18:
Seems as if another driver was installed for the tablet, most likely a generic windows 10 driver. Try checking in device manager for the tablet and then uninstall the driver then try running TabletDriver again without restarting

Kottowo, 2019-05-18:
when i tried that, it gave me an error that it could not find the tablet

Kottowo, 2019-05-18:
when i re plugged it in, it works again, but it's the same issue. The thing that I am getting is that it says this (look at the screenshot below) but i have tried uninstalling the wacom drivers multiple times 
[image]

InfinityGhost, 2019-05-18:
Do they stay on your system after uninstalling? Make sure the Wacom tablet service isn't running in `services.msc`

Kottowo, 2019-05-18:
wacom professional services is running

Kottowo, 2019-05-18:
i stopped it, restarted the driver but it still says the same thing (like in the screen above)

Kottowo, 2019-05-18:
even when i just restarted my pc, it immediately works (without drivers)

Kottowo, 2019-05-18:
[image]
should i try uninstalling the wacom tablet from my installed programs?

Kottowo, 2019-05-18:
well, that was the whole issue, me not deleting the "wacom tablet"
still ty for the effort you gave in the hopes of fixing this

closed 2019-05-18 by Kottowo