hawku/TabletDriver #419: Huion Kamvas 13 Pro

kool601, 2019-05-20:
I tried to make my own config to no avail. Hope somebody can either help or make one.

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: V0.2.3

Tablet model: Huion Kamvas Pro 13/GT-133

InfinityGhost, 2019-05-22:
What was the last state of your config? Somewhere to start is always helpful

kool601, 2019-05-22:
I know I was doing it completely wrong, but here's my config file appended with a .txt for uploadability (not a word but whatever XD).
[tablet.cfg.txt](https://github.com/hawku/TabletDriver/files/3209645/tablet.cfg.txt)

InfinityGhost, 2019-05-23:
Does it detect it as the new tablet or does it still throw "Tablet not found" or detect it as another tablet?

kool601, 2019-05-23:
It detects it, but there's no mouse cursor movement or clicking.

InfinityGhost, 2019-05-23:
Try this config
[tablet.cfg](https://github.com/hawku/TabletDriver/files/3213880/tablet.cfg.txt)

kool601, 2019-05-24:
The pen nib makes the screen go to the desktop on press. Other than that, nothing, no cursor movement.

SasquatchLord123, 2021-04-22:
Hey I am trying to create a config for my kamvas pro 12 gt-116 tablet and was wondering how you create a config and where to find the data. Thx a lot and any help is appreciated