hawku/TabletDriver #420: Driver not working

vensosa, 2019-05-23:
Tablet Driver isn't working 
This is what comes up in the console:
[2019-05-23 15:22:50.998] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2019-05-23 15:22:50.999] [ERROR] Can't open VMulti device!

[2019-05-23 15:22:50.999] [ERROR] Possible fixes:
[2019-05-23 15:22:50.999] [ERROR] 1) Install VMulti driver
[2019-05-23 15:22:50.999] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2019-05-23 15:22:50.999] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
[2019-05-23 15:22:56.689] [ERROR] Can't open XP-Pen's VMulti! Trying VEIKK's VMulti!
[2019-05-23 15:22:56.690] [ERROR] Can't open VMulti device!

[2019-05-23 15:22:56.690] [ERROR] Possible fixes:
[2019-05-23 15:22:56.690] [ERROR] 1) Install VMulti driver
[2019-05-23 15:22:56.690] [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
[2019-05-23 15:22:56.690] [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!

I tried uninstalling and installing vmulti and I uninstalled my wacom driver, but I dont know where the pentabletsevice.exe is so i cannot uninstall that.
## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: cth480

dumbstersky, 2019-06-01:
same thing is happening to me i tried to do everything it asked

ricardobrug, 2019-06-27:
go to task manager, details and look for pentabletservice.exe

yolpang, 2020-06-12:
> go to task manager, details and look for pentabletservice.exe

It's not coming out. What should I do in this case?

xVexal, 2020-09-24:
Same porblem for me... i got a wacom ctl 480 and cant run the hawku driver. I cant find this pentabletservice.exe

zachuza, 2021-08-06:
did we ever get a fix for this because i have the same issue, pentabletservice is nowhere and ive uninstalled and reinstalled vmulti countless times

yolpang, 2021-08-09:
> 나는 같은 문제가 있고 pentabletservice는 어디에도 없으며 vmulti를 무수히 제거했다가 다시 설치했기 때문에 이 문제를 해결한 적이 있습니까?

Try rerouting the installation path to drive C or check for driver crashes