hawku/TabletDriver #421: Gaomon s620 driver not working

Vsyncrooo, 2019-05-24:
I have given my friend a new tablet but the drivers don't seem to be detecting it. We tried the newest release and a pre-release of the drivers without much succes. Help would be appreciated

## Hardware/software information
Windows Version: 10

Tablet Driver version: 2.3/pre2.4

Tablet model: gaomon s620

InfinityGhost, 2019-05-24:
Can you drop your log files here?

Vsyncrooo, 2019-05-25:
My friend gave me this

[2019-05-25 00:29:12.981] Log file 'startuplog.txt' opened.
[2019-05-25 00:29:12.981] >> LogDirect true
[2019-05-25 00:29:12.981] Log direct print = True
[2019-05-25 00:29:12.981] >> Include "config\tablet.cfg"
[2019-05-25 00:29:12.981] \ Reading 'config\tablet.cfg'
[2019-05-25 00:29:12.981] >> #
[2019-05-25 00:29:12.982] >> # Example tablet definition:
[2019-05-25 00:29:12.982] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-05-25 00:29:12.982] >> #
[2019-05-25 00:29:12.982] >> # VID: 0x056a
[2019-05-25 00:29:12.982] >> # PID: 0x00dd
[2019-05-25 00:29:12.982] >> # HID Usage Page: 0x0D
[2019-05-25 00:29:12.982] >> # HID Usage: 0x01
[2019-05-25 00:29:12.982] >> #
[2019-05-25 00:29:12.982] >> #
[2019-05-25 00:29:12.982] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-05-25 00:29:12.982] >> #
[2019-05-25 00:29:12.982] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-05-25 00:29:12.987] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-05-25 00:29:12.998] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-05-25 00:29:12.998] >> ReportId 0x02
[2019-05-25 00:29:12.998] >> ReportLength 11
[2019-05-25 00:29:12.998] >> DetectMask 0x40
[2019-05-25 00:29:12.998] >> MaxX 14720
[2019-05-25 00:29:12.998] >> MaxY 9200
[2019-05-25 00:29:12.998] >> MaxPressure 1023
[2019-05-25 00:29:12.998] >> Width 147.2
[2019-05-25 00:29:12.998] >> Height 92.0
[2019-05-25 00:29:12.998] >> TabletFormat WacomDrivers
[2019-05-25 00:29:12.998] >> #
[2019-05-25 00:29:12.998] >> # Wacom CTL-470 
[2019-05-25 00:29:12.998] >> #
[2019-05-25 00:29:12.998] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-05-25 00:29:13.006] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-05-25 00:29:13.016] >> Name "Wacom CTL-470"
[2019-05-25 00:29:13.016] >> ReportId 0x02
[2019-05-25 00:29:13.016] >> ReportLength 10
[2019-05-25 00:29:13.016] >> DetectMask 0x40
[2019-05-25 00:29:13.016] >> MaxX 14720
[2019-05-25 00:29:13.016] >> MaxY 9200
[2019-05-25 00:29:13.017] >> MaxPressure 1023
[2019-05-25 00:29:13.017] >> Width 147.2
[2019-05-25 00:29:13.017] >> Height 92.0
[2019-05-25 00:29:13.017] >> InitFeature 0x02 0x02
[2019-05-25 00:29:13.017] >> #
[2019-05-25 00:29:13.017] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-05-25 00:29:13.017] >> #
[2019-05-25 00:29:13.017] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-05-25 00:29:13.024] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00

-------------------------------------------------------------------------------------------------------

hope its the right thing

InfinityGhost, 2019-05-25:
The log should be much larger than that. It is located at `TabletDriver\startuplog.txt`

Abbacus07, 2019-07-13:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3388373/startuplog.txt)

Here is my log file, I am having the same issue

Jet-chann, 2019-07-26:
I'm having the same problem, my Gaomon S620 is not getting detected,

[2019-07-25 21:50:30.060] \ Reading 'config/init.cfg'
[2019-07-25 21:50:30.060] >> Log startuplog.txt
[2019-07-25 21:50:30.060] Log file 'startuplog.txt' opened.
[2019-07-25 21:50:30.060] >> LogDirect true
[2019-07-25 21:50:30.060] Log direct print = True
[2019-07-25 21:50:30.060] >> Include "config\tablet.cfg"
[2019-07-25 21:50:30.061] \ Reading 'config\tablet.cfg'
[2019-07-25 21:50:30.061] >> #
[2019-07-25 21:50:30.061] >> # Example tablet definition:
[2019-07-25 21:50:30.061] >> #   Tablet 0x056a 0x00dd 0x0D 0x01
[2019-07-25 21:50:30.061] >> #
[2019-07-25 21:50:30.061] >> # VID: 0x056a
[2019-07-25 21:50:30.061] >> # PID: 0x00dd
[2019-07-25 21:50:30.061] >> # HID Usage Page: 0x0D
[2019-07-25 21:50:30.061] >> # HID Usage: 0x01
[2019-07-25 21:50:30.061] >> #
[2019-07-25 21:50:30.061] >> #
[2019-07-25 21:50:30.061] >> # Wacom CTL-470 (Wacom drivers installed)
[2019-07-25 21:50:30.061] >> #
[2019-07-25 21:50:30.061] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-07-25 21:50:30.065] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-07-25 21:50:30.065] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-07-25 21:50:30.065] >> ReportId 0x02
[2019-07-25 21:50:30.065] >> ReportLength 11
[2019-07-25 21:50:30.065] >> DetectMask 0x40
[2019-07-25 21:50:30.065] >> MaxX 14720
[2019-07-25 21:50:30.065] >> MaxY 9200
[2019-07-25 21:50:30.065] >> MaxPressure 1023
[2019-07-25 21:50:30.065] >> Width 147.2
[2019-07-25 21:50:30.065] >> Height 92.0
[2019-07-25 21:50:30.065] >> TabletFormat WacomDrivers
[2019-07-25 21:50:30.065] >> #
[2019-07-25 21:50:30.065] >> # Wacom CTL-470 
[2019-07-25 21:50:30.065] >> #
[2019-07-25 21:50:30.065] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-07-25 21:50:30.068] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-07-25 21:50:30.068] >> Name "Wacom CTL-470"
[2019-07-25 21:50:30.068] >> ReportId 0x02
[2019-07-25 21:50:30.068] >> ReportLength 10
[2019-07-25 21:50:30.068] >> DetectMask 0x40
[2019-07-25 21:50:30.068] >> MaxX 14720
[2019-07-25 21:50:30.068] >> MaxY 9200
[2019-07-25 21:50:30.069] >> MaxPressure 1023
[2019-07-25 21:50:30.069] >> Width 147.2
[2019-07-25 21:50:30.069] >> Height 92.0
[2019-07-25 21:50:30.069] >> InitFeature 0x02 0x02
[2019-07-25 21:50:30.069] >> #
[2019-07-25 21:50:30.069] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-07-25 21:50:30.069] >> #
[2019-07-25 21:50:30.069] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-07-25 21:50:30.072] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-07-25 21:50:30.072] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-07-25 21:50:30.072] >> ReportId 0x02
[2019-07-25 21:50:30.072] >> ReportLength 11
[2019-07-25 21:50:30.072] >> DetectMask 0x40
[2019-07-25 21:50:30.072] >> MaxX 14720
[2019-07-25 21:50:30.072] >> MaxY 9200
[2019-07-25 21:50:30.072] >> MaxPressure 1023
[2019-07-25 21:50:30.072] >> Width 147.2
[2019-07-25 21:50:30.072] >> Height 92.0
[2019-07-25 21:50:30.072] >> TabletFormat WacomDrivers
[2019-07-25 21:50:30.072] >> #
[2019-07-25 21:50:30.072] >> # Wacom CTH-470
[2019-07-25 21:50:30.072] >> #
[2019-07-25 21:50:30.072] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2019-07-25 21:50:30.076] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-07-25 21:50:30.076] >> Name "Wacom CTH-470"
[2019-07-25 21:50:30.076] >> ReportId 0x02
[... 1207 lines cut ...]
[2019-07-25 21:52:42.635]   Report Id = 0x07
[2019-07-25 21:52:42.635]   Report Length = 8 bytes
[2019-07-25 21:52:42.635]   Detect Mask = 0x80
[2019-07-25 21:52:42.635]   Ignore Mask = 0x00
[2019-07-25 21:52:42.635]   Pen button map = 'MOUSE1' 'ESCAPE' 'LCTRL+O' 
[2019-07-25 21:52:42.635]   Aux button count = 0
[2019-07-25 21:52:42.635]   Init string ids: 100 200 
[2019-07-25 21:52:42.635] 
[2019-07-25 21:52:42.635] Area:
[2019-07-25 21:52:42.635]   Desktop = 3840px x 1086px
[2019-07-25 21:52:42.635]   Screen Map = 1920px x 1080px @ X+1920px, Y+0px
[2019-07-25 21:52:42.635]   Tablet =  90.00mm x 50.00mm @ Min(80.00mm, 50.00mm) Max(170.00mm, 100.00mm)
[2019-07-25 21:52:42.635]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
[2019-07-25 21:52:42.635] 
[2019-07-25 21:52:42.635] >> Start
[2019-07-25 21:52:47.727] TabletDriver started!
[2019-07-25 21:52:47.727] [STATUS] TABLET Unknown Huion Tablet! Check startuplog.txt!
[2019-07-25 21:52:47.727] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2019-07-25 21:52:47.727] [STATUS] WIDTH 100.00000
[2019-07-25 21:52:47.727] [STATUS] HEIGHT 100.00000
[2019-07-25 21:52:47.727] [STATUS] MAX_X 10000
[2019-07-25 21:52:47.727] [STATUS] MAX_Y 10000
[2019-07-25 21:52:47.727] [STATUS] MAX_PRESSURE 2047
[2019-07-25 21:52:47.727] [STATUS] AUX_BUTTONS 0
[2019-07-25 21:52:47.727] >> Log Off

shika7200, 2019-07-26:
How to build for linux( ubuntu) ?

Azza-W, 2019-08-17:
can i get some help too? i've downloaded [this](https://github.com/InfinityGhost/TabletDriver/releases/tag/v0.2.4-pre) the first time i ran it my pen was working just fine, then afterwards it's stopped working

heres the startup log
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3511491/startuplog.txt)

InfinityGhost, 2019-08-18:
> can i get some help too? i've downloaded [this](https://github.com/InfinityGhost/TabletDriver/releases/tag/v0.2.4-pre) the first time i ran it my pen was working just fine, then afterwards it's [...]

Try a fresh copy of the release

Rinkushimo, 2019-08-20:
Im sorry but I just can't get it too work I have a s620 myself and I tried to download that pre rel version for it and it does not detect the tablet and the tablet area, screen and stuff is green whats kinda weird. How do I get it too work :( I installed the 2.3 and then the 2.4 and tried deinstalling the 2.3 version before installing the 2.4 and nothing worked :/

Abbacus07, 2019-08-22:
> Im sorry but I just can't get it too work I have a s620 myself and I tried to download that pre rel version for it and it does not detect the tablet and the tablet area, screen and stuff is green [...]

Try replacing the tablet.cfg file with the one [here](https://drive.google.com/open?id=1PlyqFpOuVmaes9X-dLbuWybC-Vs1Pqb2)

Azza-W, 2019-08-27:
> > Im sorry but I just can't get it too work I have a s620 myself and I tried to download that pre rel version for it and it does not detect the tablet and the tablet area, screen and stuff is green whats kinda weird. How do I get it too work :( I installed the 2.3 and then the 2.4 and tried deinstalling the 2.3 version before installing the 2.4 and nothing worked :/
> 
> Try replacing the tablet.cfg file with the one [here](https://drive.google.com/open?id=1PlyqFpOuVmaes9X-dLbuWybC-Vs1Pqb2)
> 
> You can find your tablet.cfg file in the tablet driver folder > config > tablet.cfg

Nah, didn't work for me i though it might've been the config file but seems not.

Azza-W, 2019-08-27:
> Try a fresh copy of the release

So that worked, but i've hit another wall here: [startuplog.txt](https://github.com/hawku/TabletDriver/files/3541629/startuplog.txt)

RevENGE-code, 2019-09-02:
here is the solution https://www.reddit.com/r/osugame/comments/cywlbt/how_to_install_hawku_driver_for_the_gaomon_s620/