hawku/TabletDriver #424: Driver not working properly

thacer13, 2019-05-29:
Tablet Driver GUI is running but behaves like its closed, tablet area size is full, clicks work differently (mouse disappears when clicking in browser and makes a sphere animation when on desktop) and cursor is stuck at corner of osu!, yesterday it was fine. I already had .NET framework, tried to repair, but a fatal error occurred (code 0x80070643), not sure if it's relevant. I also disabled Windows Ink and uninstalled all Wacom stuff, though it says (Wacom drivers installed)

Full log: 
[2019-05-29 13:27:26.704] >> #
[2019-05-29 13:27:26.704] >> HIDTablet 0x056a 0x00dd 0xFF00 0x000A
[2019-05-29 13:27:26.706] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-05-29 13:27:26.706] >> Name "Wacom CTL-470 (Wacom drivers installed)"
[2019-05-29 13:27:26.706] >> ReportId 0x02
[2019-05-29 13:27:26.706] >> ReportLength 11
[2019-05-29 13:27:26.706] >> DetectMask 0x40
[2019-05-29 13:27:26.706] >> MaxX 14720
[2019-05-29 13:27:26.706] >> MaxY 9200
[2019-05-29 13:27:26.706] >> MaxPressure 1023
[2019-05-29 13:27:26.706] >> Width 147.2
[2019-05-29 13:27:26.706] >> Height 92.0
[2019-05-29 13:27:26.706] >> TabletFormat WacomDrivers
[2019-05-29 13:27:26.706] >> #
[2019-05-29 13:27:26.706] >> # Wacom CTL-470 
[2019-05-29 13:27:26.706] >> #
[2019-05-29 13:27:26.706] >> HIDTablet 0x056a 0x00dd 0x0D 0x01
[2019-05-29 13:27:26.708] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DD UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-05-29 13:27:26.708] >> Name "Wacom CTL-470"
[2019-05-29 13:27:26.708] >> ReportId 0x02
[2019-05-29 13:27:26.708] >> ReportLength 10
[2019-05-29 13:27:26.708] >> DetectMask 0x40
[2019-05-29 13:27:26.708] >> MaxX 14720
[2019-05-29 13:27:26.708] >> MaxY 9200
[2019-05-29 13:27:26.708] >> MaxPressure 1023
[2019-05-29 13:27:26.708] >> Width 147.2
[2019-05-29 13:27:26.708] >> Height 92.0
[2019-05-29 13:27:26.708] >> InitFeature 0x02 0x02
[2019-05-29 13:27:26.708] >> #
[2019-05-29 13:27:26.708] >> # Wacom CTH-470 (Wacom drivers installed)
[2019-05-29 13:27:26.708] >> #
[2019-05-29 13:27:26.708] >> HIDTablet 0x056a 0x00de 0xFF00 0x000A
[2019-05-29 13:27:26.710] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-05-29 13:27:26.710] >> Name "Wacom CTH-470 (Wacom drivers installed)"
[2019-05-29 13:27:26.710] >> ReportId 0x02
[2019-05-29 13:27:26.710] >> ReportLength 11
[2019-05-29 13:27:26.710] >> DetectMask 0x40
[2019-05-29 13:27:26.710] >> MaxX 14720
[2019-05-29 13:27:26.710] >> MaxY 9200
[2019-05-29 13:27:26.710] >> MaxPressure 1023
[2019-05-29 13:27:26.710] >> Width 147.2
[2019-05-29 13:27:26.710] >> Height 92.0
[2019-05-29 13:27:26.710] >> TabletFormat WacomDrivers
[2019-05-29 13:27:26.710] >> #
[2019-05-29 13:27:26.710] >> # Wacom CTH-470
[2019-05-29 13:27:26.710] >> #
[2019-05-29 13:27:26.710] >> HIDTablet 0x056a 0x00de 0x0D 0x01
[2019-05-29 13:27:26.711] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DE UsagePage=0x000D Usage=0x0001 Exclusive=False
[2019-05-29 13:27:26.712] >> Name "Wacom CTH-470"
[2019-05-29 13:27:26.712] >> ReportId 0x02
[2019-05-29 13:27:26.712] >> ReportLength 10
[2019-05-29 13:27:26.712] >> DetectMask 0x40
[2019-05-29 13:27:26.712] >> MaxX 14720
[2019-05-29 13:27:26.712] >> MaxY 9200
[2019-05-29 13:27:26.712] >> MaxPressure 1023
[2019-05-29 13:27:26.712] >> Width 147.2
[2019-05-29 13:27:26.712] >> Height 92.0
[2019-05-29 13:27:26.712] >> InitFeature 0x02 0x02
[2019-05-29 13:27:26.712] >> #
[2019-05-29 13:27:26.712] >> # Wacom CTH-670 (Wacom drivers installed)
[2019-05-29 13:27:26.712] >> #
[2019-05-29 13:27:26.712] >> HIDTablet 0x056a 0x00df 0xFF00 0x000A
[2019-05-29 13:27:26.713] [WARNING] Can't open HID tablet Vendor=0x056A Product=0x00DF UsagePage=0xFF00 Usage=0x000A Exclusive=False
[2019-05-29 13:27:26.713] >> Name "Wacom CTH-670 (Wacom drivers installed)"
[2019-05-29 13:27:26.713] >> ReportId 0x02
[2019-05-29 13:27:26.714] >> ReportLength 11
[2019-05-29 13:27:26.714] >> DetectMask 0x40
[... 215 lines cut ...]
[2019-05-29 13:27:26.728] 
[2019-05-29 13:27:26.728] >> Start
[2019-05-29 13:27:26.729] TabletDriver started!
[2019-05-29 13:27:26.729] [STATUS] TABLET Wacom CTL-472 (Wacom drivers installed)
[2019-05-29 13:27:26.729] [STATUS] HID 056A 037A FF00 000A
[2019-05-29 13:27:26.729] [STATUS] WIDTH 152.00000
[2019-05-29 13:27:26.729] [STATUS] HEIGHT 95.00000
[2019-05-29 13:27:26.729] [STATUS] MAX_X 15200
[2019-05-29 13:27:26.729] [STATUS] MAX_Y 9500
[2019-05-29 13:27:26.729] [STATUS] MAX_PRESSURE 2047
[2019-05-29 13:27:26.729] [STATUS] AUX_BUTTONS 0
[2019-05-29 13:27:26.729] >> Log Off
[2019-05-29 13:27:26.733] Log file 'startuplog.txt' closed.
[2019-05-29 13:27:26.733] >> LogDirect False
[2019-05-29 13:27:26.733] Log direct print = False
[2019-05-29 13:27:26.733] 
[2019-05-29 13:27:26.733] Driver started!
[2019-05-29 13:27:26.733] 

## Hardware/software information
Windows Version: Windows 10 64-bit

Tablet Driver version: latest

Tablet model: CTL-472

InfinityGhost, 2019-05-29:
Best action is to uninstall the drivers for your tablet in **device manager**, then check for any files left behind from the wacom drivers. This seems more like a driver conflict issue here moreso than a TabletDriver issue.

thacer13, 2019-05-29:
Where? 
[Screenshot_1](https://user-images.githubusercontent.com/51170751/58583188-d7f5f000-8229-11e9-90a0-332ae15274ff.png)

InfinityGhost, 2019-05-30:
In most cases, it'll be listed under `Human Interface Devices`
[image]

closed 2019-05-30 by thacer13