hawku/TabletDriver #425: g640s cursor slides+latency

BaconPug, 2019-06-05:
whenever I stop my cursor, there is still movement like sliding that is very noticeable. I can also feel latency for my tablet. The cursor teleports a little bit when I press my buttons on the pen even though they are completely disabled. I might be doing something wrong but I don't know, I just got my new tablet and this happened

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.2.3

Tablet model: G640S

YChvanov, 2019-06-05:
When you open an issue it displays you related issues, so why didn't you check #397?
This is a hardware problem and it is not related to this driver

BaconPug, 2019-06-05:
> When you open an issue it displays you related issues, so why didn't you check #397?
> This is a hardware problem and it is not related to this driver

the problems are not the same. Though you might be right that this may be a hardware problem. It has nothing to do with anti-smoothing. There's a problem with the cursor sliding which wasn't mentioned in # 397. anti smoothing also didn't fix my delay, but I am not sure if this is the driver's problem so I am going to close this

closed 2019-06-05 by BaconPug