hawku/TabletDriver #430: support for xp pen drivers (star 03)

Saxarokkkk, 2019-06-15:
## Hardware/software information
Windows Version: win 10 pro 

Tablet Driver version: latest

Tablet model: xp pen star 03

closed 2019-06-17 by Saxarokkkk

Akiz09, 2019-06-17:
#
# XP-Pen Star 03 
#
HIDTablet 0x5543 0x0081 0xFF0A 0x0001
Name "XP-Pen Star 03"
ReportLength 8
DetectMask 0x80
MaxX 20320 //not shure
MaxY 15240 //not shure
MaxPressure 2047
Width 101.6 //not shure
Height 76.2 //not shure
InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00

TakejiReydzey, 2020-11-03:
> # XP-Pen Star 03
> HIDTablet 0x5543 0x0081 0xFF0A 0x0001
> Name "XP-Pen Star 03"
> ReportLength 8
> DetectMask 0x80
> MaxX 20320 //not shure
> MaxY 15240 //not shure
> MaxPressure 2047
> Width 101.6 //not shure
> Height 76.2 //not shure
> InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00

doesn't work for me tho